CongducPham/LowCostLoRaGw #77: does it support Stm32 end devices?

mohamedouali, 2017-02-28:
is it possible to communicate with stm32 boards with the raspberry gateway via Lora? I am using the semtech sx1278 modules for both raspberry as GW and stm32 as end device

CongducPham, 2017-02-28:
The current lib is not for STM32. Would need lots of reworking.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

mohamedouali, 2017-03-02:
thanks alot for the reply :)
But are you going to work on it?

CongducPham, 2017-03-02:
I don't think so, there are lots of lib that can work with STM32, and also the mbed initiatives that are great. We are focusing on Arduino-compatible boards with the Arduino IDE.

mohamedouali, 2017-03-02:
Ok thanks ! Just to make things clear Stm32's end devices won't work with your gateway ? 
Because I guess whatever the microcontroller on which the node is based, the LoRa packets formats won't change because of it ? Or am I wrong?

CongducPham, 2017-03-02:
>
> Ok thanks ! Just to make things clear Stm32's end devices won't work 
> with your gateway ?
If you have a lib for STM32 and use our packet format or put the gateway 
in raw format mode, then the gateway can receive from your STM32 device.
>
> Because I guess whatever the microcontroller on which the node is 
> based, the LoRa packets formats won't change because of it ? Or am I 
> wrong?
The PHY layer won't change, but we also have a small header on the LoRa 
payload.

Look at here: http://cpham.perso.univ-pau.fr/LORA/WAZIUP/FAQ.pdf

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|