CongducPham/LowCostLoRaGw #78: not receiving data on the gateway

Nuno-Nico, 2017-03-04:
I am having some trouble to receive data on the gateway, booth the arduino and the gateway seems to be working fine but no data is beeing received.

booth the arduino and the gateway was set to mode 2 , channel 11 (868)
[image]

CongducPham, 2017-03-04:
> testnw

Well, it seems that your gateway is not correctly working as the 
returned state for various configuration (such as power setting) is not 
0. It is not possible to set power above 20dBm and also for 20dBm you 
must be sure that your radio can support it. I don't know which radio 
module you are using. The preamble length is not good neither you should 
have 8. This means that your radio module has some issue. Try to get the 
latest version of the lora_gateway.cpp file here:

https://raw.githubusercontent.com/CongducPham/LowCostLoRaGw/master/gw_full_latest/lora_gateway.cpp

recompile and retry by setting power to 14dBm for instance.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Nuno-Nico, 2017-03-04:
this is strange. when i tried it now it worked, but after unplugging the arduino from the power and pluging it again, it stoped receiving again. and then after a few packets sent it star receiving again.

is it possible that the reason are radio interferences (the end-node and the gateway are just a meter apart)?

this is the print screen of it working

[image]

closed 2017-03-22 by Nuno-Nico