CongducPham/LowCostLoRaGw #79: The custom OS is not running

lucasf100, 2017-03-07:
Hi,

The boot doesnt want to boot on the Rasperry
we do exactly the same steps as described in the tutorial :/
How can we figure it out ?

Friendly
Lucas

CongducPham, 2017-03-07:
Did you get the SD card with our Raspbian image? Anyway, if the OS is not booting it is not a gateway issue but rather an OS issue. Maybe the SD card is corrupted, try with another one. You should be able to boot on the RPI even if the gateway program is not launched.

lucasf100, 2017-03-07:
We tried with an other OS (official OS) and it s working,
Its only with the "raspberrypi-jessie-WAZIUP-demo" that it doesent work, we have nothing on the screen...

lucasf100, 2017-03-08:
We did like your video on Youtube but at the end we have : "Invalid Argument"

youms, 2025-04-02:
Can anyone still have a copy of the jessie custom image?