CongducPham/LowCostLoRaGw #80: what is LAS and what is LR-BS?

mohamedouali, 2017-03-09:
what does LAS stand for and what is it actually? same question for LR-BS

mohamedouali, 2017-03-09:
I found out what is LAS which is Lora Activity sharing? what is it and what is LR BS

CongducPham, 2017-03-09:
Long-Range Base Station: the LoRa gateway if it is lora technology.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|