CongducPham/LowCostLoRaGw #81: How to send commands from web application?

mohamedouali, 2017-03-14:


mohamedouali, 2017-03-14:
what I want to say is that is it possible to use your open source work and modify it to create a gateway that sends commands to control end devices and  in the same time collect data from it with another application apart from the platforms provided like things speak ?

CongducPham, 2017-03-16:
Yes, it is up to you to do such thing. There is already a downlink 
capability that you can enhance.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|