CongducPham/LowCostLoRaGw #86: Low-power mode

namsontr, 2017-03-22:
Hi Mr Pham Cong Duc,
 I used Arduino Pro Mini in its 3.3v, 8MHz which connected the HopeRF RFM 92W (Low-cost LoRa Iot Device: A step-by-step tutorial). But when i removed the power led and the board was put in <<sleep mode>>, Wakes-up every 10 min, take a measure (temp) and send to GW 45 mA in sleep mode (expected 160uA in deep sleep mode) , 160 mA when active and sending.
Do you have any ideal this situation?
Thank you so much in advance for your valuable help.

CongducPham, 2017-03-22:
> i removed the power led and the board was put in <>, Wakes-up every 10

Which example are you compiling?

Regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

namsontr, 2017-03-22:
I'm compiling Arduino_Lora_temp & Arduino_Lora_Simple_temp. Uncomment Paboost, powerlevel='x'.
Thank you for your quickly answer.

CongducPham, 2017-03-22:
is LOW_POWER uncommented?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

namsontr, 2017-03-22:
Yes, uncomment Low_power. Successfully switch Lora module in sleep mode.
This is your code:

/*
 *  temperature sensor on analog 8 to test the LoRa gateway
 *
 *  Copyright (C) 2016 Congduc Pham, University of Pau, France
 *
 *  This program is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.

 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with the program.  If not, see <http://www.gnu.org/licenses/>.
 *
 *****************************************************************************
 * last update: Nov. 26th by C. Pham
 */
#include <SPI.h> 
// Include the SX1272
#include "SX1272.h"

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//
#define ETSI_EUROPE_REGULATION
//#define FCC_US_REGULATION
//#define SENEGAL_REGULATION
/////////////////////////////////////////////////////////////////////////////////////////////////////////// 

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
#define BAND868
//#define BAND900
//#define BAND433
///////////////////////////////////////////////////////////////////////////////////////////////////////////

#ifdef ETSI_EUROPE_REGULATION
#define MAX_DBM 14
// previous way for setting output power
 char powerLevel='x';
#elif defined SENEGAL_REGULATION
#define MAX_DBM 10
// previous way for setting output power
// 'H' is actually 6dBm, so better to use the new way to set output power
// char powerLevel='H';
#elif defined FCC_US_REGULATION
#define MAX_DBM 14
#endif

#ifdef BAND868
#ifdef SENEGAL_REGULATION
const uint32_t DEFAULT_CHANNEL=CH_04_868;
#else
const uint32_t DEFAULT_CHANNEL=CH_10_868;
#endif
#elif defined BAND900
const uint32_t DEFAULT_CHANNEL=CH_05_900;
#elif defined BAND433
const uint32_t DEFAULT_CHANNEL=CH_00_433;
#endif

[... 486 lines cut ...]
#else
          // use the delay function
          delay(LOW_POWER_PERIOD*1000);
#endif                        
          PRINT_CSTSTR("%s",".");
          FLUSHOUTPUT; 
          delay(10);                        
      }
      
      delay(50);
#endif      
      
#else
      PRINT_VALUE("%ld", nextTransmissionTime);
      PRINTLN;
      PRINT_CSTSTR("%s","Will send next value at\n");
      // use a random part also to avoid collision
      nextTransmissionTime=millis()+(unsigned long)idlePeriodInMin*60*1000+(unsigned long)random(15,60)*1000;
      PRINT_VALUE("%ld", nextTransmissionTime);
      PRINTLN;
  }
#endif

  delay(50);
}

namsontr, 2017-04-04:
I'm so sorry Mr Duc because i connected wrong pin VCC of  HopeRF RFM 92W. I  fixed this situation.

CongducPham, 2017-04-05:
Great to hear that the issue is fixed. 

Regards.

thecockroach, 2019-04-02:
Hi, I have the same problem, in sleep mode the power consumption is 5 mA. I made my own board, based on Pro Mini (3v3 8MHz), I don't understand the PABOOST define and his relation with pin DIO0, I have the SX1278 module and I'm using it with the Arduino_Lora_Generic_DHT.

CongducPham, 2019-04-02:
Check that there is no led active and that the voltage regulator is removed. See this tutorial: https://www.youtube.com/watch?v=2_VQpcCwdd8

thecockroach, 2019-04-02:
Thanks, It is the LDO the problem, now I have 18 uA in sleep Mode.