CongducPham/LowCostLoRaGw #89: Looking for help in making multiple nodes to send data to the gateway

sachethegde, 2017-04-11:
Hello, I am experimenting on ping-pong example and its working perfect. My goal is to send the data from more than one node to gateway.
Can anyone please help me how can I make multiple nodes to send pings to my gateway?

Thanks in advance.

CongducPham, 2017-04-11:
Well, I don't see the issue of having multiple nodes sending to the gw, 
could you be more specific?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

sachethegde, 2017-04-12:
Thank you for your response. 
I am still a newbie here and trying to understand how does adding an additional node work? How does my gateway differentiate between the two "ping"s coming in?
It would be really great if you can tell me where do I need to change the code.

Thanks,
Sachet