CongducPham/LowCostLoRaGw #90: LoRa vs LoRaWAN

github4f, 2017-04-16:
Hi- Could you please let me know the gatway is for LoRa or LoRaWAN? tnx

dovetion, 2017-04-17:
LoRa is a modulation technique, you can understand it as a physical layer implementation.
LoRaWAN is a protocol, a standard. It lies above the physical layer, it's a MAC layer.

github4f, 2017-04-17:
Ya I understand that point. It is well-explained for example here: http://www.libelium.com/development/waspmote/documentation/lora-vs-lorawan/
My question is that that layer (MAC) is already supported here (in this gateway) or not?

github4f, 2017-04-17:
I see your point. but I have a different point of view. I worked with dragino gateway. It also use exactly the same approach as this gateway. Singlechannel with SX127x (supporting only LORA, not LORAWAN). Now the issue is that only 20 nodes can connect to the gateway. Beyond than that, it faces considerable collision. So what counter measures can be used to tackle this issue in this gateway?    It is better buy an expensive one (250$) which can handle 1000 nodes, or installing 50 gateways (1000/20=50) which cost 50*30=1500$? I appreciate your comment.

CongducPham, 2017-04-17:
I'm a bit surprised by the limit of 20 nodes that you mention. You can 
handle much more with appropriate device randomization and also channe 
access method such as CSMA-like mechanism, which is implemented in our lib.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

github4f, 2017-04-17:
In the manual it is mentioned that "if the end nodestrytosend data to the gateway at every 5 minutes, there will be data lost after the network has 20~30nodes due to Channel Collision"! Source: http://www.dragino.com

Do you have any experience to see how many nodes can be connected to the this gateway. Assumption: every 5 minutes we send 32B data. And max distance between nodes and the gateway is around 5km. Thanks and regards.

CongducPham, 2017-04-17:
I don't know where the problem is with Dragino. I can tell you that this 
mentioned limit has not actual reasons.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Kevin-Sangeelee, 2017-04-17:
Dragino are hardware people (which they are good at), and the software is
only to show that the hardware is working. The example you referred to uses
a crude retry mechanism that is quick to discard data. Where
coding/chirp/spreading settings result in long transmissions, it's easy to
see how they could hit this limit with 30 units on a 5 minute transmit
period when using the example sketch. You would really have to modify it to
be of use in production (or use more sophisticated software, such as
LowCostLoRaGw!).

As an aside, the example uses the ThingSpeak library for Arduino, which
offers very little for the amount of microcontroller memory used. It's far
more efficient to simply form a URL string.

Kevin