CongducPham/LowCostLoRaGw #92: Bugs in post_processing_gw.py

ChiaChinHo, 2017-04-30:
Thank you for sharing this great project!

When I tried to upload data to my cloud project, I found a bug in post_processing_gw.py.
At line 845, `ldata = getAllLine()`, ldata always gets "the buffer line".
It means that _linebuf never changes. 

Maybe it needs `fillLinebuf(n)` or other functions with `sys.stdin.read()` or `sys.stdin.readline()` before `ldata = getAllLine()`.

Thank you again for this project. It helps me a lot :)
(It's my first time to new a issue, hope that I didn't do something offensive.)

CongducPham, 2017-04-30:
Hi, no you did not do something offensive.

If you receive the \ character it means that you receive from the radio, 
so the low-level gateway program has inserted the data prefix \xFF\xFE 
to indicate that data comes from the radio. As you can see at line 991, 
we read fill the buffer there, so at line 845 we read from the buffer.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ChiaChinHo, 2017-05-01:
Thanks for reply. Now, I know where my fault is.

closed 2017-05-01 by ChiaChinHo