CongducPham/LowCostLoRaGw #93: Not receiving downlinks

Benjamin1231, 2017-05-02:
Hi,

Recently started using the library and it has been wonderful so far. Unfortunately I can't make the downlink work properly. I'm running the library on a Raspberry Pi 3 with a Dragino Lora Shield. Receiving packages is going fine without any problems. However, when the server sends the package back I can't pick it up with any device.

I tried using the example from the downlink README but without any success. **The Raspberry receives the packages:**

`Starting 'availableData'
Valid Header received in LoRa mode
Checking destination
Packet received is for me

Starting 'getPacket'
Packet correctly received in LoRa mode 
Packet received:
Destination: 1
Source: 1
Packet number: 2
Packet length: 8
Data: hi`

**The Raspberry then sends this into the ether:**
`Process downlink requests 1: { "status" : "send_request", "dst" : 1, "data" : "for device 6" }

status = send_request
dst = 1
data = for device 6
--> CAD duration 201
OK1
 RSSI value is -104 
--> RSSI -104
 Packet set and written in FIFO 
 Packet to send:
Destination: 1
Type: 16
Source: 1
Packet number: 0
Packet length: 16
Data: for device 6
 

Packet sent, state 0
JSON record: {"status":"sent","dst":1,"data":"for device 6"}
`
Nothing is received on the interactive Arduino device nor on the other boards I programmed with different libraries. Does anyone know what might go wrong here?

CongducPham, 2017-05-02:
Hi, it is normal because if you use the example template (simple_temp or 
temp) you will see that the device does not open any receive window. You 
can implement it "à la" LoRaWAN with a receive window after a 
transmission. To see the downlink features, use a device with the 
interactive_device sketch.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Benjamin1231, 2017-05-02:
@CongducPham Thanks for your reply! Unfortunately I already tested it with the interactive sketch and other devices where I manually opened it after a transmission (A big window of around 25 seconds). Today I managed to make it work only once but after that couldn't get the same results. I tested it with different Dragino LoRa shields, different RX windows, power supplies, but it was all in vain. 

I have the feeling the problem lies with the TX from the gateway. I connected a multimeter to measure the current and noticed a current drop (to 6 mA) whenever the gateway tries to send something. On the other device the current rises to around 120mA when a transmission happens. The LoRa shield has it's own power supply and a common ground with the Raspberry Pi.

CongducPham, 2017-05-03:
OK, so that may come from the setting of the PA_BOOST setting on the 
radio module. Depending on your radio, radio.makefile should either have 
PABOOST or not. The Dragino uses an RFM95W so the radio.makefile should 
have CFLAGS=-DPABOOST, see here:

https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Benjamin1231, 2017-05-03:
I just tried setting the PA_BOOST setting but also without any luck. The gateway receives packages correctly but the packages sent by the gateway are not received by other devices.

Benjamin1231, 2017-05-04:
I'm also getting a receive error 2 periodically. Not sure if this is somehow related to the TX but it causes the gateway to restart every 5 minutes or so.

CongducPham, 2017-05-04:
Hi, did you get the latest version of the lib?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Benjamin1231, 2017-05-04:
@CongducPham 
Yes, I cloned the git repo according to the various tutorials on the git.

--
Benjamin

CongducPham, 2017-05-04:
OK, let me check this issue. If I understood well, this happen when you 
activate the downlink features right?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Benjamin1231, 2017-05-04:
@CongducPham,

Correct. I built the lora_gateway_pi2_downlink with PABOOST enabled and tried it with PABOOST disabled. The gateway does it's job until transmitting the LoRa package. I'm not receiving anything on either the Arduino or other devices I programmed.

Thanks for looking into it! Really appreciate that.

Benjamin1231, 2017-05-17:
Can happily say this issue was resolved now. I was using the LMIC library to send and receive data but the LoRa Sync word was different between the LMIC library and the LowCostLoRaGw. After changing this to the same Sync word I received the packages on my devices!

closed 2017-05-17 by Benjamin1231

CongducPham, 2017-05-17:
Great to hear that it worked. I didn't know that you were using the LMIC 
at the device side.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2017-05-18:
Hi, BTW, you told me that you are using LMIC, so the LoRaWAN packet format I suppose. For downlink message, our gateway currently does not use LoRaWAN format so how do you receive the packet on the end node? Or maybe you are not using LoRaWAN format with LMIC?

Benjamin1231, 2017-05-18:
Hi!

I mainly used the LMIC library because of the SX1276 drivers. My end node is a ESP32 running FreeRTOS and currently, as far as I know, there are no drivers / libraries available for an ESP32 with the RTOS environment. That's why I made a wrapper for the LMIC library so I have access to the drivers (also to the LoRaWAN protocol). For uplinks/downlinks with the Raspberry I'm just using raw packets. Looks a bit like this: [https://github.com/matthijskooijman/arduino-lmic/blob/master/examples/raw/raw.ino](url)

CongducPham, 2017-05-18:
Hi, ok, understand now, thanks for the clarification.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|