CongducPham/LowCostLoRaGw #94: ACK is not being received by end nodes.

PravallikaNagesh, 2017-05-08:
The end nodes are not being able to receive ACK from Gateway though it is able to send.
The end nodes wait for ACK after sending the data packet to Gateway for 10sec, when timeout happens it retries and again waits for ACK but still, nodes are not able to receive it.Please help me out regarding this.Thanks in advance.

CongducPham, 2017-05-09:
Hi, are you sure that the gateway is receiving? What example are you 
trying? PingPong example?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

PravallikaNagesh, 2017-05-10:
Yes, Gateway is able to receive the packets.I am working with Arduino_Lora_Simple_temp and Arduino_Lora_InteractiveDevice examples.

CongducPham, 2017-05-15:
If you request an ACK thne the gateway is going to send data, in this 
case, check how the gateway has been compiled becaseu depending on your 
radio module, you may need to set PABOOST to true for the gateway. Check 
here:

https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-27:
Hi,

  Is your issue of not receiving ack solved ? I m facing the same issue. I m able to receive ack only when gateway and node are placed near to each other. But at far distances, only the data packet is received but the node is not getting any ack from the gateway though the node is requesting ack.