CongducPham/LowCostLoRaGw #97: Gateway does not receive( or log) any packet after Ctr+C

ThanhCongVu, 2017-05-13:
Hi sir,
my gateway and end-device are working perfectly,
However, when I exit the " waiting RF input for transparent RF-serial bridge" by using Ctr+C. The incoming packets are not logged in Dropbox log files.
Am I doing something wrong by using command Ctr+C?
pls give me recommendation.

kind regards

CongducPham, 2017-05-14:
If you use CTRL-C then you killed the process. The best way to run an 
operational gateway is to reboot it. The gateway program will be 
launched on boot and will run in background. Then you can connect to the 
gateway, follow log file,...

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-06-07:
Much appreciate it Sir

closed 2017-06-07 by ThanhCongVu