CongducPham/LowCostLoRaGw #98: Issues in range for receiving message

SunRioy, 2017-05-15:
Hi Professor

I have a gateway and a node with NICERF sx1276 915MHZ. I try to test the distance that is  from the node to gateway is only 500m. I  had uncomment PA_boost and put antenna for both node and gateway. So, I am unsure what is the problem?Because the offical website for NICERF i think it is support 20dm and it can received until 3km. Do I need to change some setting in gateway to like 20dm perhaps?

Regard

CongducPham, 2017-05-15:
Well, normally the nicerf does need PABOOST, so many it is an antenna issue?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

SunRioy, 2017-05-16:
I had put a antenna for gateway and node, but still getting short distance?May i know if i use PA_BOOST for nicerf what is the dm is?and normally how far  it can go?

CongducPham, 2017-05-16:
What example are you using? In our example sketch, the default transmit 
power is 14dBm. We have LOS tests of more than 10kms.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

SunRioy, 2017-05-16:
I am using Arduino_LoRa_Simple_temp.ino sketch but the sensor I used is ultrasonic sensor, HCSR04. It is the problem of NiceRf SX1276 915MHZ cannot receive that far? I get the longest distance is 668m.