CongducPham/LowCostLoRaGw #100: Unrecognized transceiver ...

futuritymmx, 2017-05-21:
Hi,

I'm having problems getting a FALIWEI RFM95W LoRa modules to work with LowCostLoRaGw and the Raspberry Pi 2 Mobel B (Revision v1.1 2014).

https://www.aliexpress.com/item/Free-Shipping-1PCS-RFM95-20-DBM-low-power-consumption-Long-Range-wireless-transceiver-module-MHZ-frequency/32595471004.html?spm=2114.13010608.0.0.ShE7zT

I've successfully managed to get the get it working with the now deprecated single_chan_pkt_fwd github project (see working output at the end of this email): 
https://github.com/tftelkamp/single_chan_pkt_fwd

but I can't seem to get this module to work with LowCostLoRaGw.

The single_chan_pkt_fwd code worked first try by following this tutorial:
https://www.hackster.io/ChrisSamuelson/lora-raspberry-pi-single-channel-gateway-cheap-d57d36

using the following RFM95W to Raspberry Pi 2 connections:

Module 3.3v -> Pi 3.3v
Module Dio0 -> Pi CLK0 / GP4
Module MIS0 -> Pi MISO / GP9
Module MISI -> Pi MISI / GP10
Module SCK -> Pi SCLK / GP11
Module NSS -> Pi GP22
Module RESET -> PI GP21/27
Module GND -> Pi GND

The module is detected as a SX1276 module and successfully receives packets, but unfortunately the code to forward received packets to The Things Network no longer works and the code is no longer being maintained.

So I've cloned the LowCostLoRaGw project and wired up the module as described here: 
http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html

I then have the following hardware configuration:

Module 3.3v -> Pi 3.3v
Module MIS0 -> Pi MISO / GP9
Module MISI -> Pi MISI / GP10
Module SCK -> Pi SCLK / GP11
Module NSS -> Pi GP8 / CE0
Module GND -> Pi GND

I compiled the project succesfully with:

make lora_gateway_pi2

and although it runs, I get the message:
Unrecognized transceiver 
...
...
... {repeat forever}
 
I read that it can help to connect the modules RESET pin to GP17 but this didn't make any difference.  

I've moved the connections back to the positions for single_chan_pkt_fwd project code and that still seems to work, receiving packets.

If anyone could help me resolve this problem I'd really appreciate it.

Below is the output from the working single_chan_pkt_fwd project code in case it helps?

pi@raspberrypi:~/lora/single_chan_pkt_fwd $ sudo ./single_chan_pkt_fwd 
SX1276 detected, starting.
Gateway ID: b8:27:eb:ff:ff:b9:ba:f6
Listening at SF7 on 868.100000 Mhz.
------------------
stat update: {"stat":{"time":"2017-05-21 15:01:41 GMT","lati":0.00000,"long":0.00000,"alti":0,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"","desc":""}}
stat update: {"stat":{"time":"2017-05-21 15:02:11 GMT","lati":0.00000,"long":0.00000,"alti":0,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"","desc":""}}
stat update: {"stat":{"time":"2017-05-21 15:08:11 GMT","lati":0.00000,"long":0.00000,"alti":0,"rxnb":0,"rxok":0,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"","desc":""}}
Packet RSSI: -111, RSSI: -111, SNR: -11, Length: 107
rxpk update: {"rxpk":[{"tmst":534370938,"chan":0,"rfch":0,"freq":868.100000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":-11,"rssi":-111,"size":107,"data":"q52hhUQSV0ySv8Wo9SIB+o4rv6EtaukV24eSGSQ7XmkJle3mdfDBFE/BRUSs9EgKu/X/+jwc9VNGwtTxlqohhtRw753OQeHlRiL5tdjvLkWAcMTIlOBZranX93E9Y+TfQsm31LM/9Km65g4="}]}
stat update: {"stat":{"time":"2017-05-21 15:08:41 GMT","lati":0.00000,"long":0.00000,"alti":0,"rxnb":1,"rxok":1,"rxfw":0,"ackr":0.0,"dwnb":0,"txnb":0,"pfrm":"Single Channel Gateway","mail":"","desc":""}}

Kind regards

Neil

CongducPham, 2017-05-22:
Well, the connections look fine, so for the moment I have no clues for 
this issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

futuritymmx, 2017-05-23:
Thanks for getting back to me.  

Is there a way to turn on some form of debugging to diagnose this module further?

Thanks
Neil

CongducPham, 2017-05-23:
Yes, by turning in SX1272.h the SX1272_debug_mode to 2 for instance, but it won't really help as it seems to get stuck at the very beginning.

futuritymmx, 2017-05-25:
Thanks again for your help.  I'll see what I can find if anything in the debug.