CongducPham/LowCostLoRaGw #104: what's the difference between RSSI & RSSIpkt?

dovetion, 2017-05-28:
hey, this question may seems quite dumb, but I still want to ask.
When I test the whole system I received those records from the gateway terminal

```
--> CAD duration 201
OK1
--> RSSI -124
^$ACK requested by 8
--- rxlora. dst=1 type=0x18 src=8 seq=58 len=10 SNR=-13 RSSIpkt=-69 BW=125 CR=4/5 SF=12
^p1,24,8,58,10,-13,-69
^r125,5,12
^t2017-05-28T09:37:40.905

--> CAD duration 201
OK1
--> RSSI -126
^$ACK requested by 8
--- rxlora. dst=1 type=0x18 src=8 seq=6 len=10 SNR=7 RSSIpkt=-87 BW=125 CR=4/5 SF=12
^p1,24,8,6,10,7,-87
^r125,5,12
^t2017-05-28T09:51:56.598

```

there's big gap between RSSI & RSSIpkt
so,kind of confused.
Thanks!

Kevin-Sangeelee, 2017-05-28:
From the datasheet (e.g. SX1272 section 5.5.5), the RSSI value is the
instantaneous signal strength, and the packet RSSI is the average signal
strength of the latest packet received. The raw register values are used to
calculate the corresponding signal strengths expressed as dBm.

So, basically, one is 'signal strength right now', and the other is 'signal
strength while receiving the packet'. The former might hint at how noisy
your environment is, the latter lets you know how clearly the packet was
received.

dovetion, 2017-05-28:
Thank you!
That helps a lot!