CongducPham/LowCostLoRaGw #106: Receive error 2

mohamedouali, 2017-06-01:
why am I receiving receive error 2

CongducPham, 2017-06-02:
sometime there are some radio module timeout, it is normally very 
unfrequent.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2017-07-05:
try getting the latest version of lora_gateway.cpp and SX1272.cpp and SX1272.h as we increased the receive window time-out.