CongducPham/LowCostLoRaGw #107: Unstable gateway

shaniastewart, 2017-06-13:
Hello,

I am having problems with my gateway that I have made from a RPI3 and an InAir9 radio module on the 900 MHz band. My end-node uses an Arduino pro mini and it sends packets, but the gateway will only sometimes receive them properly (usually only on the first packet). I have ensured that the PABOOST is not enabled and that both modes and channels are the same for both devices, so I am not sure what could be wrong.

On the temp example, my node will show this: 

LoRa temperature sensor, extended version
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 254
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 239
Temp is 77.02
Sending \!#1#TC/77.2
Real payload size is 12
--> CAD duration 276
OK1
--> waiting for 1 CAD = 31
--> CAD duration 275
OK2
LoRa pkt size 16
LoRa pkt seq 254
LoRa Sent in 907
LoRa Sent w/CAD in 7657
Packet sent, state 0
Switch to power saving mode
Successfully switch LoRa module in sleep mode

But the gateway only shows:

2017-06-13T10:43:57.019060> **********Power ON: state 0
2017-06-13T10:43:57.019234> Default sync word: 0x12
2017-06-13T10:43:57.019400> LoRa mode 2
2017-06-13T10:43:57.019556> Setting mode: state 0
2017-06-13T10:43:57.019711> Channel CH_12_868: state 0
2017-06-13T10:43:57.019865> Set LoRa power dBm to 14
2017-06-13T10:43:57.020018> Power: state 0
2017-06-13T10:43:57.020169> Get Preamble Length: state 0
2017-06-13T10:43:57.020322> Preamble Length: 8
2017-06-13T10:43:57.020477> LoRa addr 1: state 0
2017-06-13T10:43:57.020630> SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
2017-06-13T10:43:57.352690> Low-level gw status ON​

CongducPham, 2017-06-13:
So are using mode 2 on the gw, the end device is also in mode 2? Also, 
you actually use 868 band, not 900, is it the same setting on both sides?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

shaniastewart, 2017-06-13:
Thank you for getting back to me so quick!

Yes, both are on mode 2. The end-node is running on the 900 MHz band and for the gateway I set it to the 900 band through the global_conf.json file, does it need to be done somewhere else as well?

CongducPham, 2017-06-13:
How do you set the band, because on the gateway output I see:

2017-06-13T10:43:57.019711> Channel CH_12_868: state 0

which is not 900 band.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

shaniastewart, 2017-06-13:
Initially I just changed freq in gateway_conf.json, but I looked again and I think I found the problem. I didn't set it to band 900 in lora_gateway.cpp on the RPI. I changed that now and it works! Thanks for your help!

closed 2017-06-15 by shaniastewart