CongducPham/LowCostLoRaGw #109: Node not connecting to gateway automatically after getting disconnected

akshyvi, 2017-06-20:
I am facing an issue where when keeping the gateway at a static position and moving the node away so after a certain range the node disconnects and of course there no payload sent from node side.Now, when I bring the node back in range it does not connect back to the network automatically created by Gateway, I have to manually reset the node to make it connect to the gateway again.When I say I got the node in the range, I mean I also tried keeping it right beside the gateway but it does not reconnect on its own.

Also, I using InAIR9  at 877MHZ.May I also know what is the maximum distance you have tested for the Lora devices at the line of sight and non-line of sight?
Please help
Thank you

CongducPham, 2017-06-20:
I think you don't understand the way it works. The node sends every 10 
minutes for instance. When reset, it will send immediatly and then every 
10 minutes. There is no such issue of what you called "disconnection".

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

akshyvi, 2017-06-20:
No, I have edited your code and set it to send instances after 1 minute period still the issue remains the same.What do you think ?

CongducPham, 2017-06-20:
As I said, even if you set the period to 1 minutes, there is some device randomization process making the time between 2 transmission to be between 1 and 3 minutes. Do you mean that when the device is close to the gw, there is nothing received even after several minutes? regards,

akshyvi, 2017-06-20:
Okay even after consideration of device randomization you can completely ignore the time period part for now since I have tested it for a longer period of time and result is the same it does not rejoin the network on its own after disconnection.Yes, I also tried keeping the node very close to the gw but it did not work.

Also, I would like to remind you that it does connect to gw on the first start but the issue is when it disconnects due to being out range or too many obstacles in between the gw and node or any other reason etc,It does not connect back on its own to the gw(I have waited for a long time so clear outs the timing issue, tried doing it by keeping near to gw so that clear outs the distance issue).
What do you think might be the problem?
Thank you for helping me out.
Regards,