CongducPham/LowCostLoRaGw #110: Not receiving after receive error 2

Benjamin1231, 2017-06-20:
Hi,

So the up and downlinks are fully working. However, when I leave my gateway on for some time I eventually get a receive error 2, after which the gateway doesn't receive anything anymore until I reboot it manually.

I tried it with different hardware like Raspberry 3 and 1B but it's giving me the same results. Using a Dragino LoRa Shield (SX1276) with the Raspberry. Any idea how I can solve this problem?

Kind regards,

Benjamin Meijejr

CongducPham, 2017-06-21:
A receive error 2 is reported as unexpected radio error. The code now 
can track for that error and is restarting the radio module. They should 
not be frequent and the gateway should start over quite nicely, although 
we haven't tested it with the Dragino extensively. How often did they 
appear? Does the gw stuck each time?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

Benjamin1231, 2017-06-21:
Hi,

Thanks for getting back to me. I get the receive error after a few minutes of being 'online'. It differs per time but the last few times it happened at around 20 minutes. The gateway seems to restart fine after getting the receive error:

2017-06-20T13:19:54.692908> Receive error 2                                                                             
2017-06-20T13:19:54.702545> Resetting radio module                                                                      
2017-06-20T13:19:54.707290> SX1276 detected, starting.                                                                  
2017-06-20T13:19:54.714409> SX1276 LF/HF calibration                                                                    
2017-06-20T13:19:54.721899> ...                                                                                         
2017-06-20T13:19:54.727658> Setting power ON: state 0                                                                   
2017-06-20T13:19:54.732256> LoRa CR 5: state 0                                                                          
2017-06-20T13:19:54.736643> LoRa SF 7: state 0                                                                          
2017-06-20T13:19:54.740881> LoRa BW 250: state 0                                                                        
2017-06-20T13:19:54.746131> Frequency 865.200000: state 0  
2017-06-20T13:19:54.750359> Use PA_BOOST amplifier line                        
2017-06-20T13:19:54.754656> Set LoRa power dBm to 14                                                                    2017-06-20T13:19:54.758776> Power: state 0                                                                              
2017-06-20T13:19:54.762976> Get Preamble Length: state 0                                                                2017-06-20T13:19:54.767031> Preamble Length: 8                                                                          
2017-06-20T13:19:54.770994> LoRa addr 1: state 0                                                                        
2017-06-20T13:19:54.775131> SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge            
2017-06-20T13:19:54.779217> Low-level gw status ON

So this happened every time to me so far with a Raspberry Pi 1 & 3. The gateway is starting at boot of Raspberry Pi.

Kind regards,

Benjamin