CongducPham/LowCostLoRaGw #111: raspberrypi-jessie-WAZIUP-demo.dmg.zip corrupt

wolsty7, 2017-06-21:
if you test the archive with 7zip or try to open the dmg with etcher it finds the file as corrupt.

CongducPham, 2017-06-21:
Hi, did you test it with the current one on the web? Because some days 
ago this issue has been solved, or at least I thought it was solved.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

wolsty7, 2017-06-21:
yes, I downloaded it a few hours ago

CongducPham, 2017-06-21:
Strange, i've did it and there is no problem with unzip.

wolsty7, 2017-06-21:
[image]
Same for 2 separately downloaded copies on w7 and w10, 7-zip 16.04

CongducPham, 2017-06-21:
> image
> <https://user-images.githubusercontent.com/14244187/27385049-8e73ff12-56d4-11e7-8686-623ee07bf580.png>
> Same for 2 separately downloaded copies on w7 and w10, 7-zip 16.04

Well, can you try from a Linux machine? I don't know what to tell you, 
I've downloaded and unzipped without any issue. I will try from a window 
machine.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

wolsty7, 2017-06-21:
it extracts correctly on ubuntu 12.04

CongducPham, 2017-06-21:
Good to hear, strange that Windows does have some issues with zip.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

daniel-reporte, 2017-06-25:
Same problem when I try to convert the dmg to iso file. I wonder if there are an ISO image in order to make a SD from windows. Thanks

hnaeem07, 2017-06-25:
coz you are using windows 32bit system, it supports zip 64 bit version.

daniel-reporte, 2017-06-25:
thanks hnaeeem07, I am using win 10 64 bit, I tried with 7zip 64, and else on Ubuntu, same error, file corrupted. I wonder why?

daniel-reporte, 2017-06-26:
wolsty 7, could you indicate me how do you extract the dmg in Ubuntu. no matter wich method I use the file seems to be corrupted. Any help? thank in advance

wolsty7, 2017-06-27:
i just extracted using the default zip manager, i got a crc64 of: 4C2A355701BFAA3E for the .dmg file both etcher and dmg2iso say it is corrupt but TransMac burn the sd-card correctly and the copy of rasbarian seems to work lol

TheChapu, 2017-07-05:
Linux kali 4.9.0-kali4-amd64 #1 SMP Debian 4.9.30-1kali1 (2017-06-06) x86_64 GNU/Linux

[image]

TheChapu, 2017-07-05:
dmg2img 
Linux kali 4.9.0-kali4-amd64 #1 SMP Debian 4.9.30-1kali1 (2017-06-06) x86_64 GNU/Linux

[image]

CongducPham, 2017-07-05:
> screenshot from 2017-07-05 11-05-07

try renaming in .img or .iso as it is not actually a true .dmg file but 
rather an .iso file.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

TheChapu, 2017-07-05:
Although Eatcher scores a "12%", it works OK.
Thank you
Later I will do the tests with a Dragino Hat

[image]
[image]

youms, 2025-04-02:
Can anyone still have a copy of the jessie custom image?