CongducPham/LowCostLoRaGw #113: Cannot receive acknowledgement from gateway

durvesh09, 2017-06-21:
Hi,

  My arduino node is able to request ACK to gateway but the gateway is not sending any ACK packet to node.

The gateway just setACK() and print the header but nothing is received back at node side.

CongducPham, 2017-06-21:
Well, I've tested it and it should work. However, when you request an 
ack from the gateway then you have to make sure that the PABOOST setting 
for the gw is correctly done for the radio module that you use. See 
https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-21:
Yes we have tried setting the PABOOST for SX1272 module. But no effect.
The "ping" is received at the gateway end, also the ACK is set and written to FIFO, but the node is not able to receive "pong" from gateway using ping pong example

CongducPham, 2017-06-21:
Well, it is working with my config. What is your radio module on the 
gateway?

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-21:
> 310069454>,

Hi,

  The Radio module on gateway is SX1272 from RadioJitter.

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

durvesh09, 2017-06-21:
The pins are MOSI,MISO,SCLK,NSS,GND,VCC

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

CongducPham, 2017-06-21:
Are you using a Raspberry or an Arduino gateway? How did you set the 
PABOOST on the gateway?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-21:
> 310071064>,
> 310072906>,

We are using Raspberry Pi gateway. We tried setting PABOOOST by
uncommenting #define PABOOST in lora_gateway.cpp, by
adding writeRegister(0x09,0x8F); in SX1272.cpp . But it didn't work.

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

CongducPham, 2017-06-21:
Hi, did you read here: 
https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

On the raspberry, you have to edit radio.makefile to set PABOOST. Please 
don't add extra code.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-21:
> gateway-program
> issues/113#issuecomment-
> 310073665>,
> 310075251>,

The radio makefile is already configured with DPABOOST .

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

CongducPham, 2017-06-21:
So if it is not working, try to not use PABOOST.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-21:
> 310078597>,
> 310094333>,

OK. i will try that. Can you share your radio makefile for reference ?
And is this problem is only due to PABOOST ? or some other ?

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

CongducPham, 2017-06-21:
Well, mine also depends on the radio module. On PA_BOOST wired module I 
have PABOOST defined, on others (RFO line) I have it commented.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-21:
> 310099395>,
> 310099988>,

mine current radio makefile is as below:

#options are: -DPABOOST -DMAX_DBM=v
#
#CFLAGS=-DPABOOST -DMAX_DBM=14

please check whether it is correct or not ?

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

durvesh09, 2017-06-21:
> m/LowCostLoRaGw/issues/113#
> AWmZISsY82ZTr-8R-

Do i need to add anything to it ?

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

CongducPham, 2017-06-21:
No, compile your gateway and retry

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-22:
No effect on recompiling gateway.
Also i m receiving "Receive error 1" on gateway

durvesh09, 2017-06-23:
Hi,

 thanks for the support.  The gateway is now working for me.

closed 2017-06-23 by durvesh09

fxsh1301, 2017-06-26:
@durvesh09 
Hi I got some problem here.
I'm using Pi3 as the gateway and Arduino pro mini as node, both of use sx1276 module.
Both of message seems worked, shows that sx1276 is detected. 
But the  node keep sending Ping but no Pong back from gateway.
The gateway boot on chip and use option5 to see the information but no msg coming.
I tried to uncomment the PABOOST in both of side ,but still no effect.
Do you know what kind of problem is it ?

durvesh09, 2017-06-26:
Hi,
   I just changed on OS .It was my os fault. After changing the raspbian OS everything just working fine.

fxsh1301, 2017-06-26:
Im using the OS that the owner provided, is yours same to me ?

Yi-Siang Ciou, B.Eng.

Taiwan, R.O.C.

E-mail: fxsh13002@gmail.com

Line ID :fxsh1301

durvesh09, 2017-06-26:
no i m using my separate raspbian jessie

fxsh1301, 2017-06-26:
Do you mean install LowCostGw on the Jessie from Raspberry official web rather than using encapsulated OS from Github?

durvesh09, 2017-06-26:
Yes correct.

fxsh1301, 2017-06-26:
@durvesh09  
Thanks mate!
I'll try it. Thanks a lot!!!

durvesh09, 2017-06-26:
sure..let me know if it works for you also.

fxsh1301, 2017-06-29:
@durvesh09 
I'm looking for the way that install LowCostGW on blank Jessie.
But it all need to install the zipped SD card.
How do you install it ?