CongducPham/LowCostLoRaGw #114: About Lora gateway on Pi3

fxsh1301, 2017-06-23:
Hi sir ! First, I'm appreciate that you share the wonderful resource.
But I'm got some little issue here about my Lora gateway on my Pi3.
The way I used is using the OS you provide and execute ./lora_gateway to run it.
The Lora module I used is Sx1276 which brought in my country, it worked with your Ping Pong example and other source code.
But my gateway not always detect my Lora module, it shows Unrecognized module sometimes.
Or it showed Error 1 and then stop working. Here's the history.

pi@raspberrypi:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 33032
^$Bad Preamble Length: set back to 8^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1
^$Receive error 1

And here's the Arduino serial port information, it did not receive ack from gateway.But surely recognize the module.

Simple LoRa ping-pong with the gateway
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
--> CAD duration 550
OK1
--> waiting for 1 CAD = 62
--> CAD duration 549
OK2
--> RSSI -128
Sending Ping
wait for ACK
no ACK
Packet sent, state 3
No Pong!

What kind of problem may be? Thanks！

CongducPham, 2017-06-23:
If you are using our OS image, then on boot, the gateway is already 
running an instance of the lora_gateway.cpp program, so actually, it 
seems that you are running another one. See here:

https://github.com/CongducPham/LowCostLoRaGw/tree/master/gw_full_latest#use-cmdsh-to-interact-with-the-gateway

"IMPORTANT NOTICE: Do not launch a new gateway instance with an existing 
one as there will be conflict on the SPI bus."

it might be your issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

fxsh1301, 2017-06-23:
Thanks for reply!
The console showed like this below:
The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Fri Jun 23 10:39:14 2017 from sam_2
=======================================* Gateway 00000027EB5A71F7 *===
0- sudo python start_gw.py & ; disown %1                             +
1- sudo ./lora_gateway --mode 1                                      +
2- sudo ./lora_gateway --mode 1 | python post_processing_gw.py       +
3- ps aux | grep -e start_gw -e lora_gateway -e post_proc -e log_gw  +
4- tail --line=25 ../Dropbox/LoRa-test/post-processing_*.log         +
5- tail --line=25 -f ../Dropbox/LoRa-test/post-processing_*.log      +
6- less ../Dropbox/LoRa-test/post-processing_*.log                   +
------------------------------------------------------* Bluetooth *--+
a- run: sudo hciconfig hci0 piscan                                   +
b- run: sudo python rfcomm-server.py                                 +
c- run: nohup sudo python rfcomm-server.py -bg > rfcomm.log &        +
d- run: ps aux | grep rfcomm                                         +
e- run: tail -f rfcomm.log                                           +
---------------------------------------------------* Connectivity *--+
f- test: ping www.univ-pau.fr                                        +
--------------------------------------------------* Filtering msg *--+
l- List LoRa reception indications                                   +
m- List radio module reset indications                               +
n- List boot indications                                             +
o- List post-processing status                                       +
p- List low-level gateway status                                     +
--------------------------------------------------* Configuration *--+
A- show gateway_conf.json                                            +
B- edit gateway_conf.json                                            +
C- show clouds.json                                                  +
D- edit clouds.json                                                  +
-----------------------------------------------------------* kill *--+
K- kill all gateway related processes                                +
k- kill rfcomm-server process                                        +
R- reboot gateway                                                    +
S- shutdown gateway                                                  +
---------------------------------------------------------------------+
Q- quit                                                              +
======================================================
Enter your choice:

Then I went to the number 1 to start the gateway.
Is there any step that I get wrong?

CongducPham, 2017-06-23:
This is what I told you, on boot, the gateway is already running an instance of the lora_gateway.cpp program, so using option 1 runs another instance which makes conflict on the SPI bus.

Just boot your gateway, and use option 5 to visualize messages received by the gateway.

Otherwise, use option K to kill all gateway process, and then you can start a new one manually.

fxsh1301, 2017-06-23:
Sorry! I got it now but still get "Pong" back to my Arduino node.
Could be my Lora module's problem ?
[image]

CongducPham, 2017-06-23:
Maybe your radio module on the gateway needs PABOOST setting. Check the radio.makefile on the raspberry and uncomment the line, then try again. Please read here: https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

fxsh1301, 2017-06-24:
Hi sir.
I uncomment the PABOOST in lora_gateway.cpp.
[image]
Changed 14 to 10 in my radio.makefile.
[image]

$ make lora_gateway
$ ./lora_gateway
It's still the same..

CongducPham, 2017-06-24:
So check also the device side. Maybe it need PABOOST also.

fxsh1301, 2017-06-24:
[image]
I uncomment the PABOOST in Ping_Pong code, but no pong back.

closed 2017-06-26 by fxsh1301