CongducPham/LowCostLoRaGw #115: Cannot run Simple temp or ping pong with arduino pro mini

durvesh09, 2017-06-23:
Hi,

  I m not able to run the node on arduino pro mini. The nodes are working perfect with arduino mega.
On arduino pro mini it just keeps saying "Arduino pro mini detected" and then "Unrecognized transceiver"
help...

CongducPham, 2017-06-23:
On the pro mini, the cs pin is pin 10:

"The program has been tested on Arduino Uno, Mega2560, Nano, Pro Mini, 
Mini, Due, Zero. We also tested on the Teensy3.1/3.2 and the Ideetron 
Nexus. The SX1272 lib has been modified to change the SPI_SS pin from 2 
to 10 when you compile for the Pro Mini, Mini (Nexus), Nano or Teensy."

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-23:
Thank you. It worked.

closed 2017-06-23 by durvesh09