CongducPham/LowCostLoRaGw #118: Gateway can't receive any msg from node

fxsh1301, 2017-06-26:
Hi sir. I'm using Pi3 to be the gateway and use the image you provided.
Using Arduino pro mini as the node with sx1276.
The code runs successful, could detect the module.
But the gateway seems can't receive msg from node.
Because I can't see any income on my Pi3, and no Ack to the node.
So the Ping_Pong example always missing the Pong.

Enter your choice:
5
----------------------------------------------------------------------
BEGIN OUTPUT
Following last lines of ../Dropbox/LoRa-test/post-processing_00000027EB5A71F7.log. CTRL-C to return
Current UTC date is
Mon 26 Jun 08:40:56 UTC 2017
2017-06-26T10:40:50.461411> Use DHT22 to get gateway temperature and humidity level
2017-06-26T10:40:50.465906> PIGPIO: the deamon is launched
2017-06-26T10:40:50.466222> post_processing_gw.py found an alert_conf section
2017-06-26T10:40:50.466531> Parsing cloud declarations
2017-06-26T10:40:50.466786> [u'python CloudMongoDB.py']
2017-06-26T10:40:50.466999> [u'python CloudMongoDB.py', u'python CloudThingSpeak.py']
2017-06-26T10:40:50.467193> Parsed all cloud declarations
2017-06-26T10:40:50.467360> post_processing_gw.py got cloud list:
2017-06-26T10:40:50.467542> [u'python CloudMongoDB.py', u'python CloudThingSpeak.py']
2017-06-26T10:40:50.467841> Parsing cloud declarations
2017-06-26T10:40:50.468249> Parsed all cloud declarations
2017-06-26T10:40:50.469115> post_processing_gw.py got encrypted cloud list:
2017-06-26T10:40:50.470168> []
2017-06-26T10:40:50.471285> Parsing cloud declarations
2017-06-26T10:40:50.472298> Parsed all cloud declarations
2017-06-26T10:40:50.472688> post_processing_gw.py got LoRaWAN encrypted cloud list:
2017-06-26T10:40:50.472910> []
2017-06-26T10:40:50.473086> Starting thread to measure gateway temperature
2017-06-26T10:40:50.473281> Starting thread to report gw status
2017-06-26T10:40:50.473468> Getting gateway temperature
2017-06-26T10:40:50.473658> 2017-06-26 10:40:50.457915
2017-06-26T10:40:50.473865> post status: gw ON, lat my_lat long my_long
2017-06-26T10:40:51.459426>
2017-06-26T10:40:51.459698> Current working directory: /home/pi/lora_gateway
2017-06-26T10:40:53.526866> Gateway TC : -999.00 C | HU : -999.00 % at 2017-06-26 10:40:53.478165

Simple LoRa ping-pong with the gateway
Arduino Pro Mini detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
--> CAD duration 550
OK1
--> waiting for 1 CAD = 62
--> CAD duration 549
OK2
--> RSSI -127
Sending Ping
wait for ACK
no ACK
Packet sent, state 3
No Pong!

CongducPham, 2017-06-26:
I don't know exactly where is the problem here. If you check everything and tried both with and without PABOOST, at both side (with same setting of course), then I don't see the issue. I've tested again the code and the ping-pong should work.

fxsh1301, 2017-06-26:
@CongducPham 
Thanks for reply.
I'm not sure that what problem is. 
Becuase both of gateway and end-device have detect the sx1276.
Here's the modules I bought with sx1276 and SPI interface.
It runs well on another code with pure Arduino.
[image]
[image]

closed 2017-06-29 by fxsh1301