CongducPham/LowCostLoRaGw #119: Unrecognized transceiver

samciou, 2017-06-27:
Hi thanks for sharing the source code.
I'm tying to test the PingPong example with my sx1276 module.
The gateway could detect and seems run OK.
But my Arduino UNO can't recognize the Lora module.
So I test it with my Arduino pro mini. It worked! 
What kind of problem it is?  I check the wire, 4 SPI ,VCC ,ground.
The Uno with module runs great with another source code.
I'm kind of confuse.

CongducPham, 2017-06-27:
The CS pin on the uno is 2, while it is 10 on the pro mini (all small 
format boards actually).

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

samciou, 2017-06-28:
@CongducPham 
Thanks! It worked !
But another issue is the node can't get ACK.
The gateway did receive Ping and send Pong back.

2017-06-28T04:00:45.606053> ## ACK set and written in FIFO ##
2017-06-28T04:00:45.606361> ## ACK to send:
2017-06-28T04:00:45.606526> Destination: 8
2017-06-28T04:00:45.606682> Source: 1
2017-06-28T04:00:45.606838> ACK number: 155
2017-06-28T04:00:45.606992> ACK length: 2
2017-06-28T04:00:45.607145> ACK payload: 0
2017-06-28T04:00:45.607301> ACK SNR last rcv pkt: 7
2017-06-28T04:00:45.607457>  ##
2017-06-28T04:00:45.607608>
2017-06-28T04:00:45.607757> --> CAD duration 201
2017-06-28T04:00:45.607912> OK1
2017-06-28T04:00:45.608067> --> RSSI -124
2017-06-28T04:00:45.608219> ACK requested by 8
2017-06-28T04:00:45.608377> --- rxlora. dst=1 type=0x18 src=8 seq=155 len=4 SNR=7 RSSIpkt=-115 BW=125 CR=4/5 SF=12
2017-06-28T04:00:45.608557> 2017-06-28T04:00:45.608265
2017-06-28T04:00:45.608714> rcv ctrl pkt info (^p): 1,24,8,155,4,7,-115
2017-06-28T04:00:45.608877> splitted in:  [1, 24, 8, 155, 4, 7, -115]
2017-06-28T04:00:45.609038> (dst=1 type=0x18(DATA ACK_REQ) src=8 seq=155 len=4 SNR=7 RSSI=-115)
2017-06-28T04:00:45.609207> rcv ctrl radio info (^r): 125,5,12
2017-06-28T04:00:45.609364> splitted in:  [125, 5, 12]
2017-06-28T04:00:45.609520> (BW=125 CR=5 SF=12)
2017-06-28T04:00:45.609674> rcv timestamp (^t): 2017-06-28T04:00:45.607
2017-06-28T04:00:45.609837>
2017-06-28T04:00:45.609987> got first framing byte
2017-06-28T04:00:45.610143> --> got data prefix
2017-06-28T04:00:45.610297> Ping

closed 2017-06-29 by samciou