CongducPham/LowCostLoRaGw #120: No ACK

durvesh09, 2017-06-27:
Hi,

  I m able to receive ack from gateway only when node and gateway are kept near .
If the gateway is put at larger distance from nodes, the data packets are being received but the acknowledgement from gateway is not received. Please help

CongducPham, 2017-06-27:
So it is either the PABOOST issue or antenna issue. Check both.

durvesh09, 2017-06-27:
I have define PABOOST as 14dbm in radio makefile.
Is there any other parameter that needs to be set ?

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

durvesh09, 2017-06-27:
For the same module the one which i m currently using , long range was
achieved using unmodified libelium library. But not working with the
modified version of the libelium library.

-- 
Thanks and Regards,

durvesh.bhole@afourtech.com  | Skype ID: durveshb9   | AfourTech

CongducPham, 2017-06-27:
So it means PABOOST should be disabled, i.e. commented as the original 
libray can not support PABOOST because it as primaraly intended for 
their Libelium SX1272 module.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

durvesh09, 2017-06-28:
I have tried commenting PABOOST but no effect.
Can you please tell me the procedure how to enable the PABOOST ?
Also how to send data from gateway to node ?

samciou, 2017-06-28:
I got same problem here...
Can't receive the ACK from gateway.

durvesh09, 2017-06-28:
Issue solved.
I changed the "writeRegister(RegPaDacReg, 0x84)" to "writeRegister(RegPaDacReg, 0x87)" and 
rate |= 0x1B in setMAXCurrent in SX1272.cpp
after performing the above changes, just uncomment the #define PABOOST in lora_gateway.cpp
and finally recompile  the lora_gateway.cpp then reboot

closed 2017-06-28 by durvesh09

samciou, 2017-06-28:
@durvesh09  
Do you have email or something that can contact with you ?

CongducPham, 2017-06-29:
Hi, the change you made is in setPowerDBM? I believe that you spot a 
bug. I've corrected it as follows:

     if (_needPABOOST) {
         // and then set the high output power config with register 
REG_PA_DAC
         writeRegister(RegPaDacReg, 0x87);
     }
     else {
         // disable high power output in all other cases
         writeRegister(RegPaDacReg, 0x84);
     }

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2017-06-29:
Well, actually, i dig back into the reference manual and the current code should be fine actually. In setPowerDBM, we only handle power less or equal to 14dbm, so  writeRegister(RegPaDacReg, 0x84); is fine in all case. If your module needs PA_BOOST then it will use the PA_BOOST pin. It is only when you need 20dBm power that  writeRegister(RegPaDacReg, 0x87); is needed. In this case, use setPower('X') to set to 20dBm. I actually added the following lines to setPowerDBM:

if (dbm == 20) {
	return setPower('X');
}

just before:
	
if (dbm > 14)
	return state;