CongducPham/LowCostLoRaGw #121: Unrecognized Transreceiver

randr97, 2017-06-28:
Im using arduino dragino shield v1.4. { no dio pins used}.
Its showing unrecognized trans-receiver. Where as the LoRa GPS hat is working fine on the RPi. 
I'm using arduino uno with dragino Shield v1.4.
Thanks

CongducPham, 2017-06-28:
On the Dragino shield, CS pin is hard wired to pin 10, so change in the 
SX1272.h file, the last #define statement (marked in red). Initial value 
is 2, for Uno format board. Change to 10.

#if defined ARDUINO_AVR_PRO || defined ARDUINO_AVR_NANO || defined 
ARDUINO_AVR_MICRO || defined ARDUINO_AVR_MINI \
|| defined __MK20DX256__ || defined __MKL26Z64__
#define SX1272_SS 10
#elif defined ARDUINO_AVR_FEATHER32U4 || defined ARDUINO_SAMD_FEATHER_M0
// on the Adafruit Feather, the RFM95W is embeded and CS pin is normally 
on pin 8
#define SX1272_SS 8
#undef SX1272_RST
#define SX1272_RST  4
#else
#define SX1272_SS 10
#endif

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

randr97, 2017-06-28:
Tried the edit ...... still problem exists........also tried un-commenting #define PABOOST in the arduino .ino {ping pong file}. Still didn't work. Also tried the edit mentioned in dragino page. http://wiki.dragino.com/index.php?title=Lora_Shield this also didn't work.
Thanks

CongducPham, 2017-06-29:
Well, then I don't know. regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|