CongducPham/LowCostLoRaGw #122: MQTT support in Gateway side

durvesh09, 2017-06-29:
Hi,

  How to add mqtt support at gateway side ? I want to send the data to my mqtt broker instead of sending it over free cloud.

CongducPham, 2017-06-29:
You have to install mqtt and probably write some code to handle it in a 
new cloud python script.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

CongducPham, 2017-11-29:
We have updated the distribution and we fully support MQTT clouds. Look at CloudMQTT.py.
regards,

matomback, 2018-06-21:
Hi Congduc,

I have the same question as above, hoping you can address this further!

I have successfully sent data to the "test.mosquito.org" cloud server. However when I create a new cloud profile and replace the server address "test.mosquito.org" in the key_MQTT.py script with a local MQTT broker address "192.168.1.146" (the system where my mosquitto mqtt broker is running) I get an error and the connection times out. I have tried including the port in the address "192.168.1.146:1883" as well as removing the "https://" beginning string to the server address within the CloudMQTT.py script. 

This unfortunately has not fixed it and I have not found anything else to try. If you have any other information or another fix it would be greatly appreciated.

Thanks for your help.

CongducPham, 2018-06-21:
It might be a problem from your server settings. I don't see any obvious reason why it would not work with another MQTT server than test.mosquitto.org. We also tried with broker.mqttdashboard.com and it works.
regards,