CongducPham/LowCostLoRaGw #124: Using SX1278 Modules with this Library and Tutorials

mamsoudi, 2017-07-03:
Hey there,
Thanks for your great efforts,

Unfortunately we can't find SX1272 modules that you've specified in [here](http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html) (Modtronix inAir 9b) in Iran, but we found this module which uses SX1278:
https://www.smart-prototyping.com/LoRa-SX1278-10KM-433M-long-range-wireless-module-Ra-01

I have two questions now:
1. Is it possible to build a gateway using your tutorials and libraries in the tutorial above and this module? since it doesn't support CSAM antennas and works in different frequency (433MHz).
2. Does this affect the range of connection or does it reduces the connectivity range?
3. If we use this module instead of the ones you've introduced we might run into problems configuring and using it or we can just follow the tutorials?

CongducPham, 2017-07-03:
Hi, should be fine. For running in 433, at the gateway, just mention 
"freq": 433.3 in gateway_conf.json and in the example sketch choose 
BAND433 as follows:

// please uncomment only 1 choice
//#define BAND868
//#define BAND900
#define BAND433

I believe the module needs PABOOST settings, so just read here:

https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

and

https://github.com/CongducPham/LowCostLoRaGw#connect-a-radio-module-to-your-end-device

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

mamsoudi, 2017-07-03:
@CongducPham Great thanks so much!

closed 2017-07-03 by mamsoudi