CongducPham/LowCostLoRaGw #127: STM32 + SX1272 as end device

nthiers, 2017-07-10:
Hi,

I'm using your RPi image for low cost gateway. It boots fine and I would like to test communication with a end device. 
I bought some [ST LoRa demo boards](http://www.st.com/en/evaluation-tools/p-nucleo-lrwan1.html) which are basically a STM32L0+SX1272 and ST is providing a [library](http://www.st.com/content/st_com/en/products/embedded-software/mcus-embedded-software/stm32-embedded-software/stm32cube-expansion-software/i-cube-lrwan.html).
I used two demo boards with "PingPong" example (LoRa mode only, no LoRaWAN) and it works fine.
If I understand correctly, the gateway starts in mode 1 (BW=125 kHz, CR=4/5, SF=12) on channel 10.
I set the same parameters on my node but I see nothing on gateway.

Are you working on LoRa mode or FSK/OOK mode?
I'm a bit confused because, from my understanding, bandwidth, coding rate, etc are parameters related to LoRa mode and node addr seems to be related to FSK mode (page 91 of [SX1272 datasheet](http://www.semtech.com/images/datasheet/sx1272.pdf)).
And in your Arduino sketches, you are setting both parameters.

Could you give some clarification?

Thanks for all the job done here, impressive.

CongducPham, 2017-07-10:
Even if LoRa is the same at physical layer, there may be some headers 
that are specific to some library. We have our own header. If you want 
the gateway to not interpret any header, then you have to start it with 
the raw option in gateway_conf.json, or simply for testing:

sudo ./lora_gateway --mode 1 --raw

also, be sure that your st device are on the same frequency.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

nthiers, 2017-07-11:
Hi,

It was it, there was no header (section 3 of your [doc](http://cpham.perso.univ-pau.fr/LORA/RPIgateway.html)) before my payload.

However, when I'm turning on a second end node (different node addr and payload), gateway displays "Receive error 2", resets the radio module, restarts gw, starts displaying one or two received packets and displays error again.
This problem disappears when I'm turning off second end node.

Do you have an idea?
Thanks for your support.

CongducPham, 2017-07-12:
Hi, can you take the last version of lora_gateway.cpp and SX1272.cpp and 
SX1272.h as we increased the received window time to avoid this type of 
error. Please keep me informed.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

JordanSilverman, 2017-10-27:
Hi,

I have been using Pham's code for a while now using arduinos as end-devices and it works very nicely. I am now switching over to using STM32F072 for the production phase. I see that you have been able to get the example code working on an STM32 using the ST libraries.

Could you guide me through the alterations that you in Pham's example code to work with the ST libraries?

I am going through a steep learning curve to understand how to use and program the STM32 microcontrollers.

Thanks in advance!