CongducPham/LowCostLoRaGw #128: Setting channels for 915MHz SX1276 on raspberrypi gateway

ai6sm3596ru61u04, 2017-07-14:
Hi, 

I am using 900MHz SX1276 transceiver but I don't know where to modify the program in raspberrypi. Could someone tell me which programming file I should look into in order to change the frequency range?

ankitnehra, 2017-07-20:
lora_gateway.cpp on line 223 comment #define BAND868 and uncomment BAND900

CongducPham, 2017-07-27:
Hi, the README has been updated. Please read:

https://github.com/CongducPham/LowCostLoRaGw#running-in-433mhz-and-900mhz-band

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

DuaneKaufman, 2017-08-14:
Hello,
I too am trying to use the Dragino HAT on a RaspberryPi3 for the 915MHz bad (US)

The HAT is marked for 915MHz use, and I have jumpered the correct pins.
I modified the lora_gateway.cpp on line 223 comment #define BAND868 and uncomment BAND900, but when I start the program (using sudo ./lora_gateway) I get:

pi@LORA_Gateway:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON

...which doesn't look like I am enabling the 915MHz channels?

Sincerely,
Duane

ankitnehra, 2017-08-14:
Did you compile it after making the change?

DuaneKaufman, 2017-08-14:
Hi ankitnehra,
Thanks for the response.
Yes, I did a make lora_gateway, then deleted the old one and linked lora_gateway_pi2 to lora_gateway, like the original installer did.

Duane

DuaneKaufman, 2017-08-15:
Hi all,
I'm sorry, I mis-interpreted the linking of lora_gateway and lora_gateway_pi2, and was running a stale copy

I now get:

pi@LORA_Gateway:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_5_900: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON

Which looks right, right?

Sincerely,
Duane

GARYARIES, 2020-11-28:
If you enter from your LAN network as http://192.168.xxx.xxx/admin
with username: admin and password: loragateway
In the gateway configuration screen you can change the working frequency