CongducPham/LowCostLoRaGw #129: Sensor data not reading

ankitnehra, 2017-07-20:
Hi,
I was using the Arduino_Lora_Ping_Pong.ino to send sensor data.
So I read data from the DHT22 sensor and instead of sending the "Ping" string, I'm sending the data read by the sensor after converting it to a string. However, the data read is always NAN. When I read the data using a different sketch (just sensor reading), it works fine.
I am not using the Generic Sensor sketch because whenever I try to use it, the setup just repeats.
Also, Generic Sensor doesn't have support for one of the sensors I am using, namely, Adafruit TCS34725.
Any help is appreciated.

CongducPham, 2017-07-21:
Well, it seems to be a problem in how you read the DHT22 value. So you 
can use GenericSensor and adapt it to your need.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ankitnehra, 2017-07-25:
I tried using Generic Sensor and was still unable to read data. I am using the Dragino LoRa shield for Arduino. Here is the Arduino ping pong code to which I made some modifications in order to include sensor data. 

When I read and print data in setup before the module is turned on using sx1272.ON(), the data is read correctly. However, once the module is turned on the digital pin doesn't seem to read anything and all I get as a reading is "nan".

Do you think there could be a wiring issue with the Dragino shield? I am reproducing the relevant code here: 

`#define DEFAULT_DEST_ADDR 1

int loraMode=LORAMODE;

//Libraries for sensors
#include <Wire.h>
#include <DHT.h>
#include <Adafruit_TCS34725.h>

//Constants
#define DHTPIN 5     // what pin we're connected to
#define DHTTYPE DHT22   // DHT 22  (AM2302)
DHT dht(DHTPIN, DHTTYPE); //// Initialize DHT sensor for normal 16mhz Arduino

/* Initialise with specific int time and gain values */
Adafruit_TCS34725 tcs = Adafruit_TCS34725(TCS34725_INTEGRATIONTIME_700MS, TCS34725_GAIN_1X);

void setup()
{
  int e;
  
  // Open serial communications and wait for port to open:
#if defined __SAMD21G18A__ && not defined ARDUINO_SAMD_FEATHER_M0 
  SerialUSB.begin(38400);
#else
  Serial.begin(38400);  
#endif 

//start up the sensors
  if (tcs.begin()) {
    Serial.println("Found sensor");
  } else {
    Serial.println("No TCS34725 found ... check your connections");
  } 
  dht.begin();

  float humi, tempe;
  humi = dht.readHumidity();
  tempe= dht.readTemperature();
  Serial.println(humi);
  Serial.println(tempe);  

  
  // Print a start message
  PRINT_CSTSTR("%s","Simple LoRa ping-pong with the gateway\n");  

#ifdef ARDUINO_AVR_PRO
  PRINT_CSTSTR("%s","Arduino Pro Mini detected\n");
#endif

#ifdef ARDUINO_AVR_NANO
  PRINT_CSTSTR("%s","Arduino Nano detected\n");
#endif

#ifdef ARDUINO_AVR_MINI
  PRINT_CSTSTR("%s","Arduino MINI/Nexus detected\n");
#endif

#ifdef __MK20DX256__
  PRINT_CSTSTR("%s","Teensy31/32 detected\n");
#endif

#ifdef __SAMD21G18A__ 
  PRINT_CSTSTR("%s","Arduino M0/Zero detected\n");
#endif

  // Power ON the module
  sx1272.ON();
  
  // Set transmission mode and print the result
  e = sx1272.setMode(loraMode);
  PRINT_CSTSTR("%s","Setting Mode: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;

  // enable carrier sense
  sx1272._enableCarrierSense=true;
    
  // Select frequency channel
  e = sx1272.setChannel(DEFAULT_CHANNEL);
  PRINT_CSTSTR("%s","Setting Channel: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;
  
  // Select amplifier line; PABOOST or RFO
#ifdef PABOOST
  sx1272._needPABOOST=true;
  // previous way for setting output power
  // powerLevel='x';
#else
  // previous way for setting output power
  // powerLevel='M';  
#endif

  // previous way for setting output power
  // e = sx1272.setPower(powerLevel); 

  e = sx1272.setPowerDBM((uint8_t)MAX_DBM); 
  PRINT_CSTSTR("%s","Setting Power: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;
  
  // Set the node address and print the result
  e = sx1272.setNodeAddress(node_addr);
  PRINT_CSTSTR("%s","Setting node addr: state ");
  PRINT_VALUE("%d", e);
  PRINTLN;
  
  // Print a success message
  PRINT_CSTSTR("%s","SX1272 successfully configured\n");

  delay(500);
}

void loop(void)
{

  uint8_t message[500];

  uint8_t r_size;
  int e;

  float hum;  //Stores humidity value
  float temp; //Stores temperature value

  String hum_str, temp_str, r_str, g_str, b_str, msg;
  
  //Light sensor
  uint16_t r, g, b, c, colorTemp, lux;

  sx1272.CarrierSense();

  sx1272.setPacketType(PKT_TYPE_DATA);

  //r_size=sprintf((char*)message, "Ping");
      
  while (1) {
      //Read data and store it to variables hum and temp
      hum = (dht.readHumidity()*100);
      temp= (dht.readTemperature()*100);
      Serial.println(hum, temp);              
`

ankitnehra, 2017-07-25:
Quick update:
It reads the data even if I put it at the very end of setup, the sensor still reads the data fine. But as soon as I try to read the data in the loop, the sensor doesn't work.

CongducPham, 2017-07-27:
If at one moment you can read the data correctly then it means that it is an issue independant from the communication framework. Try to debug your code further, by lokking for instance at how your local variable are instanciated/used in the loop.

regards,

Kevin-Sangeelee, 2017-07-27:
I think your use of Serial.println() is wrong. The second parameter is used
to specify the output format. You have passed in the value for temperature.
Try outputting humidity and temperature separately.

Regarding the actual reading from the DHT22, it's worth noting that the
Dragino LoRa shield has a number of jumpers that can be used to connect DIO
pins of the Semtech chip to the Arduino GPIO pins. If one of these pins
conflicts with the pin that you use for the DHT22, then it could affect the
sensor's data output at instants where the DIO pin is not in a high
impedance state. You can check the docs/schematic for the shield to find
out the pin routing.