CongducPham/LowCostLoRaGw #131: Error in post_processing_gw.py at line 896, 'ascii' codec can't decode...

vincentgosselin1, 2017-08-01:
Hi,

sometimes my gateway cannot transmit to my cloud because of a UnicodeDecodeError. This only happens once in a while. 

The error is : 'ascii' codec can't decode byte 0xb2 in position 49: ordinal not in range(128)

I'm linking also a screenshot I took when this error happens in gateway.
[image]

Thank you so much!

-Vincent

CongducPham, 2017-08-01:
> gw_error

Hi, I see that you are using ruby so you have a ruby script, can this 
error come from your ruby script?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

vincentgosselin1, 2017-08-01:
Hi,
I believe the error is in Python because the Ruby script does not get fired yet.
Here is a successful transmission when the Ruby script gets fired:
[image]

Thanks!

-Vincent

CongducPham, 2017-08-01:
Hi, could you send me your version of post_processing_gw.py please as 
mine may not be the one you got.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

vincentgosselin1, 2017-08-01:
Hi,

I cloned this entire Github on June 10th 2017 and simply added .replace(' ','') to ldata on line 879 to remove spaces. The error was already happening before that modification.

Here it is: 
[post_processing_gw.py.zip](https://github.com/CongducPham/LowCostLoRaGw/files/1192008/post_processing_gw.py.zip)

Thanks!

-Vincent

CongducPham, 2017-08-01:
What may happen is a corrupted payload. Try using AES encryption as with AES we also use the LoRaWAN-like MIC integrity checking to discard corrupted payload. The fact is that we did observe some strange reception payload when the SNR is very low, e.g. -22 for instance. However we expected that the radio module would reject the packet due to bad CRC but it was not the case and we still haven't spotted why. However in your case, the SNR is still very good.