CongducPham/LowCostLoRaGw #132: ACK receive message

elgebar777, 2017-08-08:
Hi, i have problem with connection between Raspberry and Arduino. When i set configuration raw on true
 ("gateway_conf": {.. "raw": true,) i get messege from Arduino, but Rasp does not sent ACK to Arduino. When i set RAW on false, my Rasp don't recognize any message - i didn't see "got first framing byte"

CongducPham, 2017-08-08:
Why are you using the raw option? It is only used in very specific 
cases. What do you want to do exactly?

regards

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

elgebar777, 2017-08-08:
I wish send string from Arduino to my Rasp with ACK, but Rasp dont send back any information.
When I use raw:true mu Rasp can recognize mess from Arduino

CongducPham, 2017-08-08:
You don't have to use raw mode, please set it to false. Then look at the 
ping ping example for sending data and requesting ack.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kkalbaugh, 2017-09-12:
I am having the same issue.   I am using a Feather M0 RFM95 and a Hope RFM95W on a RasPI 3 B.   The M0 sends the "Ping" and the RasPI gateway receives the PING but the M0 never receives back the "PONG".   

The message from the RasPi Gateway:
```
--> CAD duration 202
OK1
--> RSSI -128
^$ACK requested by 8
--- rxlora. dst=1 type=0x18 src=8 seq=102 len=4 SNR=6 RSSIpkt=-61 BW=125 CR=4/5 SF=12
^p1,24,8,102,4,6,-61
^r125,5,12
^t2017-09-12T09:33:38.004
▒▒Ping
## ACK set and written in FIFO ##
## ACK to send:
Destination: 8
Source: 1
ACK number: 103
ACK length: 2
ACK payload: 0
ACK SNR last rcv pkt: 6
```

Message from the M0:
```
Sending Ping
wait for ACK
no ACK
Packet sent, state 3
No Pong!
```

Thanks,
Kirby

CongducPham, 2017-09-12:
When the gateway is sending an ack, you have to make sure that at the 
gateway, the lora_gateway program is compiled with the correct PABOOST 
settings that is needed by the radio module. Please read:

https://github.com/CongducPham/LowCostLoRaGw#compiling-the-low-level-gateway-program

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kkalbaugh, 2017-09-12:
C. Pham,

Thanks for replying.   I have this in the radio.makefile:

```
#options are: -DPABOOST -DMAX_DBM=v
#
CFLAGS=-DPABOOST -DMAX_DBM=14
```
I get this:

`make lora_gateway_pi2`

```
g++ -DPABOOST -DMAX_DBM=14 -DRASPBERRY -DRASPBERRY2 -DIS_RCV_GATEWAY -c lora_gateway.cpp -o lora_gateway_pi2.o
g++ -lrt -lpthread lora_gateway_pi2.o arduPi_pi2.o SX1272_pi2.o -o lora_gateway_pi2
rm -f lora_gateway
ln -s lora_gateway_pi2 ./lora_gateway
```

I don't know if this matters but when I start the gateway I get a message that says:
`./lora_gateway: unrecognized option '--ndl'`

kkalbaugh, 2017-09-12:
I should have included the entire start up message:

```
Running interactively the full LoRa gateway... CTRL-C to exit
sudo ./lora_gateway --bw 125 --cr 5 --sf 12 --ch 10 --ndl | python post_processing_gw.py | python log_gw.py
./lora_gateway: unrecognized option '--ndl'
post_processing_gw.py found an alert_conf section
Parsing cloud declarations
[u'python CloudMongoDB.py']
Parsed all cloud declarations
post_processing_gw.py got cloud list:
[u'python CloudMongoDB.py']
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got encrypted cloud list:
[]
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got LoRaWAN encrypted cloud list:
[]
Starting thread to report gw status
2017-09-12 12:51:30.252501
post status: gw ON, lat my_lat long my_long

Current working directory: /home/pi/lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
**********Power ON: state 0
Default sync word: 0x12
LoRa CR 5: state 0
LoRa SF 12: state 0
LoRa BW 125: state 0
Channel CH_10_900: state 0
Use PA_BOOST amplifier line
Set LoRa power dBm to 14
Power: state 0
Get Preamble Length: state 0
Preamble Length: 8
LoRa addr 1: state 0
SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Low-level gw status ON
```

This time I ran it via option 0 (cmd.sh) but I get the same result if it loads on boot.

CongducPham, 2017-09-12:
>
> C. Pham,
>
> Thanks for replying. I have this in the radio.makefile:
>
> |#options are: -DPABOOST -DMAX_DBM=v # CFLAGS=-DPABOOST -DMAX_DBM=14|
|What is your radio module?|
> ||
>
> I get this:
>
> |make lora_gateway_pi2|
>
> |g++ -DPABOOST -DMAX_DBM=14 -DRASPBERRY -DRASPBERRY2 -DIS_RCV_GATEWAY
> -c lora_gateway.cpp -o lora_gateway_pi2.o g++ -lrt -lpthread
> lora_gateway_pi2.o [...]
|No it doesn't matter|

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

kkalbaugh, 2017-09-12:
HopeRF RFM95W Transceiver Module.

CongducPham, 2017-09-12:
So PABOOST setting is fine. If you have this setting and you don't 
receive the ack on the device then I don't know what is the issue.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

mefy85, 2017-09-14:
@CongducPham ,
Hi, I have a similar problem with my gateway, Raspberry pi 3 + Adafruit RFM95W (868Mhz)... the arduino test client (ping pong) with the same radio module, correctly send the ping and in the stdout I clear read the packet with rssi -61 dbm, the gateway send the ack but arduino recives that only in a few meters range... when I get far nothing happen... I think is the PABOOST of gateway, so I compile the radio.makefile but ... nothing... How can I check if the PABOOST is on or off?
thank you in advance!
and greetings for your great job!
Michele

CongducPham, 2017-09-14:
What do mean by "so I compile the radio.makefile"? You have to uncomment 
the line CFLAGS=-DPABOOST, save the file and then run

make lora_gateway_pi2

again.

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|