CongducPham/LowCostLoRaGw #133: Can't get downlink packet from GW

Povolnov, 2017-08-12:
Hi, first of all big thank you for you work and really great project. Everything fine, but i stuck with one issue.

I use  Arduino Mega (Sketch Interactive) + NiceRF1276 transceivers. Can get ACK packets, Ping-Pong work fine as well. But I can't receive Downlink message. 

--- Arduino side ---
Rcv serial: give me a downlink
Sending. Length is 18
give me a downlink
--> CAD duration 183
OK1
--> waiting for 3 CAD = 186
--> CAD duration 183
OK2
--> RSSI -122
Packet number 7
Payload size is 18
ToA is w/4B header 1614
LoRa Sent in 261
LoRa Sent w/CAD in 815
Packet sent, state 0
--- 

-- GW --
got first framing byte
--> got data prefix
give me a downlink
2017-08-12 10:56:09.972049
post downlink: checking for downlink/downlink-post.txt
post downlink: no downlink requests
post downlink: list of pending downlink requests
{ "status" : "send_request", "dst" : 6, "data" : "for device 6" }
-----------------------------------------------------
Check for downlink requests 2017-08-12T10:56:17
Read downlink 1: 66
Read downlink 2: -1
Queue all valid downlink requests
Downlink entry 1: { "status" : "send_request", "dst" : 6, "data" : "for device 6" }
status = send_request
dst = 6
data = for device 6
JSON record: {"status":"queued","dst":6,"data":"for device 6"}
-----------------------------------------------------
-----------------------------------------------------
Process downlink requests 1: { "status" : "send_request", "dst" : 6, "data" : "for device 6" }

status = send_request
dst = 6
data = for device 6
--> CAD duration 201
OK1
--> RSSI -116
Packet sent, state 0
JSON record: {"status":"sent","dst":6,"data":"for device 6"}
2017-08-12 10:56:19.982081
post downlink: checking for downlink/downlink-post.txt
post downlink: no downlink requests
---

And finally no any news from Arduino side.

Any ideas ?

Thank you in advice !

Andrey Povolnov
Russia National Science Research Institute of Communications

closed 2017-08-12 by Povolnov

jtroeth1, 2018-07-04:
@Povolnov Did you solve this? I am having the same issue...?

Thanks

CongducPham, 2018-07-05:
Try the Arduino_LoRa_temp example for downlink.