CongducPham/LowCostLoRaGw #136: Send GPS koordinates with gy-gps6mv2

bkrajnc, 2017-08-17:
I have Arduino Nano, RFM95W 868Mhz and installed  LowCostloRaGwGW and working ping-pong sketch  :-). I was try with DHT22 and it works to. How to connect GPS module gy-gps6mv2 and send coordinates to lora GW? GPS  Module have VCC, GND, TX and RX .

CongducPham, 2017-08-20:
Well, each sensor (including GPS module) has specific way for being 
wired and to communicate with the host microcontroller. You have to 
implement it yourself, however, we have this example for that works fine 
with the GY-GPS6MV2, NEO6MV2 Ublox NEO-6M Module. Hope it helps.

https://www.dropbox.com/sh/3g3a9c1fdxfn5lf/AAA5uphT0ZeBWWCQz1eYv3sVa?dl=0

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bkrajnc, 2017-08-21:
Many thanks for the information and files. I will try and will also inform about progress.

Regards

bkrajnc, 2017-08-21:
Hello
I connect my GPS module  to Arduino . There is already Lora connected (from previous test project) 

GPS VCC - > Nano +5V
GPS TX - > Nano RX
GPS RX - > Nano TX
GPS GND -> Nano GND

I upload the sketch you send, and I get output on serial:
Simple LoRa GPS sensor
Arduino Nano detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 9
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured

I am not shure if Setting Mode (and other ) is OK if there is 0 output. I assumee that there is something wrong with serial connected to GPS module. What should I change in part  ///CHANGE HERE THE GPS SERIAL PORT///

Appreciated for any advice

Regards

bkrajnc, 2017-08-26:
...heh, I find where the problem was, and actualy it was on serial connection. There is wrong labeled TX and RX on GPS Module. When I change that I get output:

$GPRMC,182930.00,V,,,,,,,260817,,,N*76
$GPVTG,,,,,,,,,N*30
$GPGGA,182930.00,,,,,0,00,99.99,,,,,,*67
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,2,1,06,03,,,31,14,,,31,25,,,22,26,,,31*7B
$GPGSV,2,2,06,31,,,30,32,,,31*7D
$GPGLL,,,,,182930.00,V,N*4B
$GPRMC,182931.00,V,,,,,,,260817,,,N*77
$GPVTG,,,,,,,,,N*30
$GPGGA,182931.00,,,,,0,00,99.99,,,,,,*66
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,1,1,04,03,,,30,14,,,31,26,,,32,32,,,31*7C
$GPGLL,,,,,182931.00,V,N*4A
$GPRMC,182932.00,V,,,,,,,260817,,,N*74
$GPVTG,,,,,,,,,N*30
$GPGGA,182932.00,,,,,0,00,99.99,,,,,,*65
$GPGSA,A,1,,,,,,,,,,,,,99.99,99.99,99.99*30
$GPGSV,1,1,04,14,,,26,2⸮⸮⸮⸮⸮⸮⸮LoRa GPS sensor
Arduino Nano detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 9
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured

Butnothing more happend from here. I change 
///////////////////////////////////////////////////////////////////
// CHANGE HERE THE TIME IN MINUTES BETWEEN 2 READING & TRANSMISSION
unsigned int idlePeriodInMin = 1;
///////////////////////////////////////////////////////////////////

That's mean,every minute data should be sent, or I am wrong ?

regards

CongducPham, 2017-08-28:
Hi, that maybe conflict with the serial port as the nano has only 1 
serial port. You can't reliably have both serial monitor and GPS on the 
same serial port. Try not to use serial monitor and look with a gateway 
to receive the GPS data. Normally, if after 3 minutes there is no GPS 
fix then the node is going back to sleep in order to retry later on. Try 
with

unsigned int idlePeriodInMin = 5;

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bkrajnc, 2017-09-01:
Thank you for your answer. I try not to use serial monitor but unfortunately nothing happend. That is all from GW output:

--> CAD duration 201
OK1
--> RSSI -124
^$ACK requested by 8
--- rxlora. dst=1 type=0x18 src=8 seq=17 len=4 SNR=6 RSSIpkt=-47 BW=125 CR=4/5 SF=12
^p1,24,8,17,4,6,-47
^r125,5,12
^t2017-09-01T09:01:44.807
▒▒Ping
 *************THIS IS OUTPUT FROM PREVIOUS PING TEST**************

Then I upload sketch for GPS

******* WAITING FOR GPS DATA*******
^$Low-level gw status ON
^$Low-level gw status ON
^$Low-level gw status ON
^$Low-level gw status ON
^$Low-level gw status ON

What do you suggest?

regards 

Boris

bkrajnc, 2017-09-08:
I try this on Mega 2560 but it doesn't work too :-(. Maybe question where I can define which serial is used for GPS module ?

Regards
Boris

StridesIT, 2018-01-30:
hello , we use a dragino shield for adrduino , the GPS is getting intitlaised and stops no data is seen on the serial monitor. we are using the above mentioned ino file -9 nov updated. 

please let us know which pins need to be connected.
[image]

CongducPham, 2018-02-05:
I havent' tried the GPS with this board yet. I don't know which serial port it is connected to, Serial0 or Serial 1? When I'm back from travel, I'll test it and get back to you.

regards,

Hiutale, 2018-02-13:
Hi

First thanks for good project.  
I have all so problem whit GPS. I use nano and NEO-m8n. Ping Pong works just fine but when upload GPS and not use serial monitor arduino not send any packets. I check GPS whit another program that fix is ok. I tried add Softwareserial, but not have skills to get it work =) So can anyone edit code for softwareserial ?

Alpha130, 2019-05-24:
>  use nano and NEO-m8n. Ping Pong works just fine but when upload GPS and not use serial monitor arduino not send any packets. I check GPS whit another program that fix is ok. I tried add Softwareserial

Comment
//#define GPS_PIN_POWER