CongducPham/LowCostLoRaGw #137: sharing PC's Internet connection through Ethernet cable

ThanhCongVu, 2017-08-18:
Hi Sir,
I have a problem with sharing PC's internet connection to Rasp Pi3 over Ethernet cable. I have seen many website showing how to do this but it does not help at all.
Is it any chance if you can make a tutorial to show us how to do it?
Currently, I am using Windows 10

Kind regards

ThanhCongVu, 2017-08-21:
Hi sir and Waziup's followers,
I found the problem which is originated from my Raspberry Pi3's faulty
For your reference on how to share Internet access from PC to RPi through Ethernet cable: I am using the method Bridging between my laptop's internet connection and the Local Area Connection.

Note that: 
make sure that you disable SHARING functionality in the PC's wireless connection before making the bridge

closed 2017-08-21 by ThanhCongVu

CongducPham, 2017-08-21:
Thank you for sharing this information. It will definitely be useful for 
windows users as we are mostly working on Linux/Mac.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|