CongducPham/LowCostLoRaGw #138: Upgrading to Raspian Stretch

kiwironnie, 2017-08-28:
Dear Prof,
The only problem (so far) that I've had upgrading to Raspian Stretch is that the compatible mongodb php driver (2.x) has changed significantly.
The earlier mongo driver (up to 1.6) can't be used as it doesn't work with the version of OpenSSL (1.1) used by Stretch.
MongoClient() is deprecated, in the new mongodb driver so index.php and export_csv.php need modifications to work.
For example rather than $connection = new MongoClient( ) to access the database requires 
$connection = new MongoDB\Driver\Manager("mongodb://localhost:27017");
It would probably take a PHP programmer a few minutes to implement the remaining required modifications, but it is a little beyond me at the moment, without several hours trying to figure out the code syntax.
Some help would be very much appreciated.   Best wishes, Ron

roens, 2018-03-03:
I also am interested in support for the newer mongo (in PHP7). I don't want to use your pre-built OS image because of need for support for another HAT (for RPi battery, power, & RTC).

I've modified `gw_full_latest/php/index.php` to use the newer connection syntax that @kiwironnie spoke of. I also prefer to use Nginx with PHP-FPM It's now failing at:

    PHP message: PHP Fatal error:  Uncaught Error: Call to a member function find() on null in /var/www/html/index.php:32
    Stack trace:
    #0 {main}
      thrown in /var/www/html/index.php on line 32" while reading response header from upstream, client: 10.0.0.154, server: _, request: "GET / HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock:", host: "loragatepi.local", referrer: "http://loragatepi.local/"
    2018/03/02 09:50:58 [error] 1414#1414: *48 FastCGI sent in stderr: "PHP message: PHP Notice:  Undefined property: MongoDB\Driver\Manager::$messages in /var/www/html/index.php on line 29
    PHP message: PHP Notice:  Trying to get property of non-object in /var/www/html/index.php on line 29
    PHP message: PHP Fatal error:  Uncaught Error: Call to a member function find() on null in /var/www/html/index.php:32
    Stack trace:
    #0 {main}
      thrown in /var/www/html/index.php on line 32" while reading response header from upstream, client: 10.0.0.154, server: _, request: "GET / HTTP/1.1", upstream: "fastcgi://unix:/var/run/php/php7.0-fpm.sock:", host: "loragatepi.local", referrer: "http://loragatepi.local/"

Any attention to upgrading this php code to more current releases of php & mongo would be very much welcome. I'm happy to try   :)

----
In the meantime, I got mine working with [`mongo-php-adapter`](https://github.com/alcaeus/mongo-php-adapter). While inside `LowCostLoRaGw/gw_full_latest/php`:

1. Install [Composer](https://getcomposer.org/) (if it's not already)
2. Create a `composer.json` file, containing just: `{}`
3. Install `mongo-php-adapter`:
`composer config "platform.ext-mongo" "1.6.16" && composer require alcaeus/mongo-php-adapter`
4. Add this line to `index.php`, above where `$connection` is defined:
`require __DIR__ . '/vendor/autoload.php';`

CongducPham, 2018-03-03:
Dear all, thank you for these feedbacks. Unfortunately the support of new MongoDB API is not planned in our schedule. If you manage to make it working, I will be happy to try to integrate these changes and update the distrib. regards,