CongducPham/LowCostLoRaGw #139: end_device configuration and serial monitor are not related

ThanhCongVu, 2017-09-09:
Hi sir,
I have configured my end_device : 
with app_key 
without AES

but in the serial monitor, I see the printed line : Encrypting and encrypted payload, which should not be printed( as I believe).
Can You please clarify this for me?
[image]
[image]

CongducPham, 2017-09-09:
> ice_screenshot_20170909-152747
> ice_screenshot_20170909-152852

That's strange, did you recompile and reflash your device?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-09-09:
certainly yes Sir

closed 2017-09-15 by ThanhCongVu