CongducPham/LowCostLoRaGw #140: end_device message format to gateway for storing in MongoDB

ThanhCongVu, 2017-09-09:
Hi sir,
I have a problem in message format from end_device to gateway to store in the MongoDB and recognising app_key.

I have tried with the original format: For example : \!#3#TC/25.12 and my_appKey[4]={5, 6, 7, 8};
where field index is 3
temperature recorded: 25.12

 it does not show in the MongoDB at the gateway( when I access to the MongoDB through web browser), however, it does show in the log file.

This is my set up:
1. in the clouds.json
I enabled the Local gateway MongoDB in clouds.json to true
but when a message received at the gateway, it shows that "number of enabled clouds is 0"

2. in the gateway_conf.json
even with app_key configured in the end_device, gateway does not show any recognition of app_key added in the message.

Have I missed anything??
[image]
[image]

hope to hear from you soon Sir

CongducPham, 2017-09-09:
> ice_screenshot_20170909-154628
> ice_screenshot_20170909-155142

Did you reboot the gateway? You can just test whether it is ok by 
killing all processes (option 3 of cmd.sh), then run

sudo python start_gw.py and see if received messages are uploaded to 
mongodb.

Also, just in case, for message to be uploaded to clouds (mongodb is 
treated as a cloud) the message must start with \!, seems that you 
forgot the \ in front of the !

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-09-10:
hi sir,
I have managed to get the app_key functionality working ( as gateway shows for example:
app key is 0x05 0x06 0x07 0x08
in app key list )
but that is all the information I got at the gateway post_processing.

Is it anyway for me to check whether the MongoDB is working or not Sir?
I have tried : sudo python CloudMongoDB.py  to check if it prints any error but nothing showed( does it mean MongoDB works fine?)

I have changed the format of the message as you mentioned above: adding \ in front of ! 

Thanks heaps Sir

ThanhCongVu, 2017-09-10:
another thing Sir,
When I access to the data from MongoDB, those data are old, which I think it is from the original image that you set up.
and in the datasheet they are dated 2016-08-04.---DATA;00000027EB27F90F;6;7;-60;5;SF12BW125;2016-08-04 14:10:31;{"TC": 22.2};
while my GW ID is 27EB246B7A so definitely it did not change the MongoDB.
can you please show me what I have done wrong at the beginning ? 
[image]

CongducPham, 2017-09-10:
> ice_screenshot_20170910-174441

Hi, is you gateway connected to the internet?

You can try the following instructions to see whether you are uploaded 
to mongodb or not.

https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-NewCloud.md#support-of-mongodb-as-a-cloud-declaration

but if you do not see something like:

|--> cloud[0] uploading with python CloudMongoDB.py .... |

Can you provide the output of your gateway when receiving a message?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-09-10:
Hi Sir,
This is my log: 
2017-09-10T21:59:11.821413> will enforce app key
2017-09-10T21:59:11.822385> post_processing_gw.py found an alert_conf section
2017-09-10T21:59:11.822816> Parsing cloud declarations
2017-09-10T21:59:11.823214> [u'python CloudMongoDB.py']
2017-09-10T21:59:11.823595> Parsed all cloud declarations
2017-09-10T21:59:11.823975> post_processing_gw.py got cloud list: 
2017-09-10T21:59:11.824358> [u'python CloudMongoDB.py']
2017-09-10T21:59:11.824740> Parsing cloud declarations
2017-09-10T21:59:11.825116> Parsed all cloud declarations
2017-09-10T21:59:11.825492> post_processing_gw.py got encrypted cloud list: 
2017-09-10T21:59:11.825877> []
2017-09-10T21:59:11.826241> Parsing cloud declarations
2017-09-10T21:59:11.826614> Parsed all cloud declarations
2017-09-10T21:59:11.826987> post_processing_gw.py got LoRaWAN encrypted cloud list: 
2017-09-10T21:59:11.827376> []
2017-09-10T21:59:11.827740> Starting thread to report gw status
2017-09-10T21:59:11.828157> 2017-09-10 21:59:11.820119
2017-09-10T21:59:11.828538> post status: gw ON, lat my_lat long my_long
2017-09-10T21:59:12.821605> 
2017-09-10T21:59:12.822131> Current working directory: /home/pi/lora_gateway
2017-09-10T21:59:12.822540> SX1276 detected, starting.
2017-09-10T21:59:12.822917> SX1276 LF/HF calibration
2017-09-10T21:59:12.823305> ...
2017-09-10T21:59:12.823667> **********Power ON: state 0
2017-09-10T21:59:12.824040> Default sync word: 0x12
2017-09-10T21:59:12.824409> LoRa mode 4
2017-09-10T21:59:12.824790> Setting mode: state 0
2017-09-10T21:59:12.825227> Frequency 913.880000: state 0
2017-09-10T21:59:12.825611> Set LoRa power dBm to 14
2017-09-10T21:59:12.825985> Power: state 0
2017-09-10T21:59:12.826352> Get Preamble Length: state 0
2017-09-10T21:59:12.826724> Preamble Length: 8
2017-09-10T21:59:12.827091> LoRa addr 1: state 0
2017-09-10T21:59:12.827460> SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
2017-09-10T21:59:13.173652> Low-level gw status ON
2017-09-10T21:59:28.094586> --- rxlora. dst=1 type=0x12 src=13 seq=206 len=17 SNR=8 RSSIpkt=-49 BW=500 CR=4/5 SF=12
2017-09-10T21:59:28.095241> 2017-09-10T21:59:28.094226
2017-09-10T21:59:28.095676> rcv ctrl pkt info (^p): 1,18,13,206,17,8,-49
2017-09-10T21:59:28.096113> splitted in:  [1, 18, 13, 206, 17, 8, -49]
2017-09-10T21:59:28.096531> (dst=1 type=0x12(DATA WAPPKEY) src=13 seq=206 len=17 SNR=8 RSSI=-49)
2017-09-10T21:59:28.096959> rcv ctrl radio info (^r): 500,5,12
2017-09-10T21:59:28.097349> splitted in:  [500, 5, 12]
2017-09-10T21:59:28.097744> (BW=500 CR=5 SF=12)
2017-09-10T21:59:28.098127> rcv timestamp (^t): 2017-09-10T21:59:28.092
2017-09-10T21:59:28.098515> 
2017-09-10T21:59:28.098896> got first framing byte
2017-09-10T21:59:28.099288> --> got data prefix
2017-09-10T21:59:28.099684> --> DATA with_appkey: read app key sequence
2017-09-10T21:59:28.100093> app key is  0x03 0x06 0x07 0x08
2017-09-10T21:59:28.100501> in app key list
2017-09-10T21:59:28.100882>

ThanhCongVu, 2017-09-10:
this is how my clouds.json looks like ( only MongoDB is enabled( true) , the rest of clouds are disable( false)
[image]

CongducPham, 2017-09-10:
Can you kill all process (option K of cmd.sh) then just run

sudo ./lora_gateway

and give me the output when receiving the message? I just want to make 
sure that you are sending the correct string sequence that triggers the 
cloud upload.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-09-10:
Hi sir,
that was the result after I used option K as kill all related processes

ThanhCongVu, 2017-09-10:
this  is the message on my end-device
[image]

ThanhCongVu, 2017-09-10:
sorry for the last comment, I did not read it carefully.
Here is the output after I use option K and then run sudo ./lora_gateway
pi@raspberrypi:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON

I just realize something strange that: I have modified the file lora_gateway.cpp as following
#define BAND900
and
#define LORAMODE 4

but in the above result after running sudo ./lora_gateway, I have a different set of configuration

ThanhCongVu, 2017-09-10:
So this is what I have done after :
make lora_gateway_pi2
----------------------------------------------------------------------------------------------------
pi@raspberrypi:~/lora_gateway $ make lora_gateway_pi2
g++  -DRASPBERRY -DRASPBERRY2 -DIS_RCV_GATEWAY -c lora_gateway.cpp -o lora_gateway_pi2.o
g++ -lrt -lpthread lora_gateway_pi2.o arduPi_pi2.o SX1272_pi2.o -o lora_gateway_pi2
rm -f lora_gateway
ln -s lora_gateway_pi2 ./lora_gateway
pi@raspberrypi:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 4
^$Setting mode: state 0
^$Channel CH_5_900: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON
--- rxlora. dst=1 type=0x12 src=13 seq=14 len=17 SNR=10 RSSIpkt=-47 BW=500 CR=4/5 SF=12
^p1,18,13,14,17,10,-47
^r500,5,12
^t2017-09-11T00:44:04.952
��   <------------------------------------------------------------------ I got this weird symbols

ThanhCongVu, 2017-09-10:
and this is what I have after using sudo ./lora_gateway | python post_processing_gw.py

pi@raspberrypi:~/lora_gateway $ sudo ./lora_gateway  | python post_processing_gw.py
will enforce app key
post_processing_gw.py found an alert_conf section
Parsing cloud declarations
[u'python CloudMongoDB.py']
Parsed all cloud declarations
post_processing_gw.py got cloud list: 
[u'python CloudMongoDB.py']
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got encrypted cloud list: 
[]
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got LoRaWAN encrypted cloud list: 
[]
Starting thread to report gw status
2017-09-11 00:48:18.313647
post status: gw ON, lat my_lat long my_long

Current working directory: /home/pi/lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
**********Power ON: state 0
Default sync word: 0x12
LoRa mode 4
Setting mode: state 0
Channel CH_5_900: state 0
Set LoRa power dBm to 14
Power: state 0
Get Preamble Length: state 0
Preamble Length: 8
LoRa addr 1: state 0
SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Low-level gw status ON
--- rxlora. dst=1 type=0x12 src=13 seq=18 len=17 SNR=10 RSSIpkt=-46 BW=500 CR=4/5 SF=12
2017-09-11T00:49:43.133569
rcv ctrl pkt info (^p): 1,18,13,18,17,10,-46
splitted in:  [1, 18, 13, 18, 17, 10, -46]
(dst=1 type=0x12(DATA WAPPKEY) src=13 seq=18 len=17 SNR=10 RSSI=-46)
rcv ctrl radio info (^r): 500,5,12
splitted in:  [500, 5, 12]
(BW=500 CR=5 SF=12)
rcv timestamp (^t): 2017-09-11T00:49:43.131

got first framing byte
--> got data prefix
--> DATA with_appkey: read app key sequence
app key is  0x03 0x06 0x07 0x08
in app key list

--- rxlora. dst=1 type=0x12 src=13 seq=19 len=17 SNR=7 RSSIpkt=-45 BW=500 CR=4/5 SF=12
2017-09-11T00:51:14.471947
rcv ctrl pkt info (^p): 1,18,13,19,17,7,-45
splitted in:  [1, 18, 13, 19, 17, 7, -45]
(dst=1 type=0x12(DATA WAPPKEY) src=13 seq=19 len=17 SNR=7 RSSI=-45)
rcv ctrl radio info (^r): 500,5,12
splitted in:  [500, 5, 12]
(BW=500 CR=5 SF=12)
rcv timestamp (^t): 2017-09-11T00:51:14.469

got first framing byte
--> got data prefix
--> DATA with_appkey: read app key sequence
app key is  0x03 0x06 0x07 0x08
in app key list

CongducPham, 2017-09-10:
So, for some reason you are not receiving the payload. Seems that you 
are only receiving the app key (the 4 strage bytes that you mentioned). 
This is reason with post-processing_gw.py is not uploading as there is 
no payload passed to its stage.

I don't know why you are not receiving the payload though. Can you send 
me the code used for your device?

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-09-10:
/*
 *  temperature sensor on analog 8 to test the LoRa gateway
 *  extended version with AES and custom Carrier Sense features
 *  
 *  Copyright (C) 2016 Congduc Pham, University of Pau, France
 *
 *  This program is free software: you can redistribute it and/or modify
 *  it under the terms of the GNU General Public License as published by
 *  the Free Software Foundation, either version 3 of the License, or
 *  (at your option) any later version.

 *  This program is distributed in the hope that it will be useful,
 *  but WITHOUT ANY WARRANTY; without even the implied warranty of
 *  MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
 *  GNU General Public License for more details.
 *
 *  You should have received a copy of the GNU General Public License
 *  along with the program.  If not, see <http://www.gnu.org/licenses/>.
 *
 *****************************************************************************
 * last update: June 17th by C. Pham
 */
#include <SPI.h> 
// Include the SX1272
#include "SX1272.h"

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//
#define ETSI_EUROPE_REGULATION
//#define FCC_US_REGULATION
//#define SENEGAL_REGULATION
/////////////////////////////////////////////////////////////////////////////////////////////////////////// 

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
// please uncomment only 1 choice
//#define BAND868
#define BAND900
//#define BAND433
///////////////////////////////////////////////////////////////////////////////////////////////////////////

#ifdef ETSI_EUROPE_REGULATION
#define MAX_DBM 14
// previous way for setting output power
// char powerLevel='M';
#elif defined SENEGAL_REGULATION
#define MAX_DBM 10
// previous way for setting output power
// 'H' is actually 6dBm, so better to use the new way to set output power
// char powerLevel='H';
#elif defined FCC_US_REGULATION
#define MAX_DBM 14
#endif

#ifdef BAND868
#ifdef SENEGAL_REGULATION
const uint32_t DEFAULT_CHANNEL=CH_04_868;
#else
const uint32_t DEFAULT_CHANNEL=CH_10_868;
#endif
#elif defined BAND900
const uint32_t DEFAULT_CHANNEL=CH_05_900;
#elif defined BAND433
const uint32_t DEFAULT_CHANNEL=CH_00_433;
#endif

// IMPORTANT
///////////////////////////////////////////////////////////////////////////////////////////////////////////
[... 879 lines cut ...]
          Snooze.deepSleep(sleep_config);
#endif  
#else
          // use the delay function
          delay(LOW_POWER_PERIOD*1000);
#endif                        
          PRINT_CSTSTR("%s",".");
          FLUSHOUTPUT; 
          delay(10);                        
      }
      
      delay(50);
#endif  
      
#else
      PRINT_VALUE("%ld", nextTransmissionTime);
      PRINTLN;
      PRINT_CSTSTR("%s","Will send next value at\n");
      // use a random part also to avoid collision
      nextTransmissionTime=millis()+(unsigned long)idlePeriodInMin*60*1000+(unsigned long)random(15,60)*1000;
      PRINT_VALUE("%ld", nextTransmissionTime);
      PRINTLN;
  }
#endif
}

ThanhCongVu, 2017-09-10:
Hi sir,
Apart from configuring the lora module( mode, channel,etc,,)
I had to change some lines due to Windows 10 Anniversary update does not let the Arduino IDE compile normally as it printed error about " sprintf ( ) " as following:

char message_app_key_offset[30]; <------------- I added this line
...
...
...
#ifdef FLOAT_TEMP
      ftoa(float_str,temp,2);
#ifdef NEW_DATA_FIELD
      r_size=sprintf( message_app_key_offset,"\\!#%d#TC/%s", field_index, float_str);  <----------This line specifically mentioned as error
...
...
...
#endif

      PRINT_CSTSTR("%s","Sending ");
      PRINT_STR("%s",message_app_key_offset); <---------- I modified this line
      PRINTLN;
      
      PRINT_CSTSTR("%s","Real payload size is ");
      PRINT_VALUE("%d", r_size);
      PRINTLN;
      
      int pl=r_size+app_key_offset;

............................................................................................................................................................

thanks sir

ThanhCongVu, 2017-09-10:
this is the output at Serial Monitor
[image]

ThanhCongVu, 2017-09-10:
Hi sir,
Apart from configuring the lora module( mode, channel,etc,,)
I had to change some lines due to Windows 10 Anniversary update does not let the Arduino IDE compile normally as it printed error about " sprintf ( ) " as following:

char message_app_key_offset[30]; <------------- I added this line
...
...
...
#ifdef FLOAT_TEMP
      ftoa(float_str,temp,2);
#ifdef NEW_DATA_FIELD
      r_size=sprintf( message_app_key_offset,"\\!#%d#TC/%s", field_index, float_str); <-------------------------------- I modified this line because the original line was pointed as error by the Arduino IDE
#else
      // this is for testing, uncomment if you just want to test, without a real temp sensor plugged
      //strcpy(float_str, "21.55567");
     // r_size=sprintf((char*)message+app_key_offset, "\\!#%d#%s", field_index, float_str);
#endif 
...
...
...
#endif

  PRINT_CSTSTR("%s","Sending ");
  PRINT_STR("%s",message_app_key_offset); <--------------------------- I modified this line
  PRINTLN;
  
  PRINT_CSTSTR("%s","Real payload size is ");
  PRINT_VALUE("%d", r_size);
  PRINTLN;
  
  int pl=r_size+app_key_offset;
............................................................................................................................................................

thanks sir

ThanhCongVu, 2017-09-10:
Hi sir, 
by some how the copy and paste does not work well on this browser, so I will send you the file
[new 1.txt](https://github.com/CongducPham/LowCostLoRaGw/files/1290650/new.1.txt)

CongducPham, 2017-09-10:
OK I see. In general, when you modify the code you have to know what you 
are doing and it seems that you do not understand the meaning of the 
app_key_offset.

In this case, you introduce an array called message_app_key_offset, fill 
it with data, but the packet that is sent is actually the content of the 
message variable, not the message_app_key_offset variable (see line 
725). So as you can see, there is only the app_key value in the message 
variable. This is why there is no payload in the packet that is received.

I don't know why there is an error with windows on the original code, 
but there is certainly a way to solve it. Maybe by using

r_size=sprintf( (char*)(&message[app_key_offset]),"\\!#%d#TC/%s", 
field_index, float_str);

instead of the original:

r_size=sprintf( (char*)message+app_key_offset,"\\!#%d#TC/%s", 
field_index, float_str);

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ThanhCongVu, 2017-09-14:
Dear Sir,
Thanks for your reply, and I apologise for modifying the code without looking at it throughfully.
In regarding to the compiling error, I still have the same problem. As I have done some research this problem is due to Windows 10 Anniversary update. So I am not sure what I should do now...
[image]

ThanhCongVu, 2017-09-15:
Hi Sir, 
I just found the solution for this so now everything is working correctly.
As a reference for those who are stuck in this situation where the Arduino IDE gives an error related to 'NO_REGS' using sprintf( ) method, You can fix the issue by downgrade the IDE to a version at least from Arduino IDE 1.6.9. 

Thanks again for helping through this sir.

closed 2017-09-15 by ThanhCongVu

CongducPham, 2017-09-15:
Actually I sent you a mail but seems that it was not received. Here it is:

---
Please check this:

"First, install the Arduino IDE. You can use the latest one (we tested with 1.8.3). But then, check (see the board manager) that the Arduino AVR board library is not above 1.6.11 as there might be some compilation issues because of the change of the GCC AVR compiler."

as explained here

https://github.com/CongducPham/LowCostLoRaGw#a-simple-end-device-example-that-periodically-sends-temperature-to-the-gateway

---

So I think that you can use up to 1.6.11

regards,

ThanhCongVu, 2017-09-15:
> that-periodically-sends-temperature-to-the-gateway

Hi sir,
I have tried Arduino IDE latest version, 1.6.10 and 1.6.11 but none of them
is working correctly.
But it works perfectly when I use 1.6.9
Thanks again sir