CongducPham/LowCostLoRaGw #143: Corrupted img file

ethicstechOSS, 2017-09-29:
I tried the latest sdcard image, http://cpham.perso.univ-pau.fr/LORA/WAZIUP/raspberrypi-jessie-WAZIUP-demo.dmg.zip

Getting corrupted error when the dmg file was extracted from the zip file. Kindly cross check it. 

TIA,
Prakash.

CongducPham, 2017-09-29:
This is an issue with 32/bit/64bit OS, try with Linux, or use Etcher to 
burn the SD card without decompressing it. regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

adool88, 2017-10-01:
Can I ask you to build new image? 7zip gives me crc error, etcher gives me long list of problems. I not sucseeded to build from scratch twice... My LoRa devices waiting for gateway.

CongducPham, 2017-10-01:
Yes, I've built a new image in the we to take into account new 
development. Will make it available monday.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ethicstechOSS, 2017-10-03:
I also tried your suggestions and confirmed that the image is not usable on the sdcard and getting similar error like adool88 reported.

CongducPham, 2017-10-03:
So you donwloaded the new version I put yesterday morning? Did you try with etcher. I've tested on my Mac and it is working fine. Do not unzip the compressed file.

adool88, 2017-10-03:
New version (v5) works fine as expected. Thanks!

CongducPham, 2017-10-03:
Great. This new image also contains the latest version of the web admin interface where update and configuration of the gateway is much easier. Check this tutorial here:
https://github.com/CongducPham/tutorials/blob/master/Low-cost-LoRa-GW-web-admin.pdf

ethicstechOSS, 2017-10-03:
I did not yet tested the new version of image. Can you share the link to download this new version of image?.

TIA,
Prakash

CongducPham, 2017-10-03:
same link:
http://cpham.perso.univ-pau.fr/LORA/WAZIUP/raspberrypi-jessie-WAZIUP-demo.dmg.zip

ethicstechOSS, 2017-10-03:
Thanks. Tested the new image. Its working.

closed 2017-10-03 by ethicstechOSS