CongducPham/LowCostLoRaGw #144: Arduino_LoRa_Generic_Sensor

bkrajnc, 2017-10-01:
I understand that something is wrong with decoding, but what ?

2017-10-01 22:26:16.159157
2017-10-01T22:26:16.160460> post status: gw ON, lat my_lat long my_long
2017-10-01T22:27:19.127396> --- rxlora. dst=1 type=0x12 src=13 seq=3 len=94 SNR=7 RSSIpkt=-40 BW=125 CR=4/5 SF=12
2017-10-01T22:27:19.128233> 2017-10-01T22:27:19.125764
2017-10-01T22:27:19.129055> rcv ctrl pkt info (^p): 1,18,13,3,94,7,-40
2017-10-01T22:27:19.129720> splitted in:  [1, 18, 13, 3, 94, 7, -40]
2017-10-01T22:27:19.130334> (dst=1 type=0x12(DATA WAPPKEY) src=13 seq=3 len=94 SNR=7 RSSI=-40)
2017-10-01T22:27:19.130950> rcv ctrl radio info (^r): 125,5,12
2017-10-01T22:27:19.131562> splitted in:  [125, 5, 12]
2017-10-01T22:27:19.132153> (BW=125 CR=5 SF=12)
2017-10-01T22:27:19.132705> rcv timestamp (^t): 2017-10-01T22:27:19.124
2017-10-01T22:27:19.133274>
2017-10-01T22:27:19.133812> got first framing byte
2017-10-01T22:27:19.134363> --> got data prefix
2017-10-01T22:27:19.134895> --> DATA with_appkey: read app key sequence
2017-10-01T22:27:19.135440> app key is  0x05 0x06 0x07 0x08
2017-10-01T22:27:19.135972> in app key list
2017-10-01T22:27:19.136503> valid app key: accept data
2017-10-01T22:27:19.137017> number of enabled clouds is 2
2017-10-01T22:27:19.137540> --> cloud[0]
2017-10-01T22:27:19.138060> uploading with python CloudMongoDB.py
2017-10-01T22:27:19.138662> 'ascii' codec can't decode byte 0xd7 in position 38: ordinal not in range(128)
2017-10-01T22:27:19.139373> --> cloud[1]
2017-10-01T22:27:19.139944> uploading with python CloudThingSpeak.py
2017-10-01T22:27:19.140489> 'ascii' codec can't decode byte 0xd7 in position 38: ordinal not in range(128)
2017-10-01T22:27:19.141072> --> cloud end

Best regards
Boris

CongducPham, 2017-10-02:
What are you sending? Can you try by launching the gateway with

sudo ./lora_gateway

to get raw output?

Don't forget to kill all gateway related process (option K of cmd.sh) 
before launching the command in a terminal.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

bkrajnc, 2017-10-07:
Hi,

Sorry for late response. I did like you said and this is the output:

pi@LoRa_GW_KR:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON
--- rxlora. dst=1 type=0x12 src=13 seq=13 len=94 SNR=7 RSSIpkt=-28 BW=125 CR=4/5 SF=12
^p1,18,13,13,94,7,-28
^r125,5,12
^t2017-10-07T22:33:59.428
▒▒ws▒▒▒▒Q~▒j▒{▒▒k▒▒0.0/DISn▒/3.50/DS/?▒{+▒Jv▒▒▒s▒▒{▒▒▒]
xterm

I try to send temperature and Humidity from DHT22 which is connected to Arduino nano. This is the output from serial on nano:

Arduino Nano detected
ATmega328P detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 13
Forced to use default parameters
Using node addr of 13
Using idle period of 10
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Sending \!tc/0.0/TC/-1.0/HU/-1.0/lw/3.50/DS/?⸮{+⸮Jv⸮⸮⸮s⸮⸮{⸮⸮⸮]
⸮⸮ws⸮⸮⸮⸮⸮⸮Q~⸮j⸮{⸮⸮k⸮⸮0.0/DISn⸮/
Real payload size is 90
--> CAD duration 548
OK1
--> waiting for 1 CAD = 62
--> CAD duration 549
OK2
LoRa pkt seq 13
LoRa Sent in 4192
LoRa Sent w/CAD in 4741
Packet sent, state 0
Switch to power saving mode
Successfully switch LoRa module in sleep mode
......................

sebaz143, 2017-10-08:
Hi, that wasn't my issue. That's somebody else thread. My issue was about
having troubles in the 900 Mhz band connecting to the gateway trough a
rn2903.

CongducPham, 2017-10-08:
You have a lot of strange characters, don't forget that ASCII format is requied, so don't write directly floating point data, you have to convert into ASCII first.

bkrajnc, 2017-10-10:
...and what should I do to convert into ASCII? Sorry for stupid questions, but I am (not so young ;-) ) beginner ....