CongducPham/LowCostLoRaGw #145: Odd values of RSSI received when SNR changes from positive to negative

richoco, 2017-10-02:
Hi, I am using two LoRa modules from RF Solutions, both paired with Arduino. The emitter has the ping pong code uploaded, and the receiver is using the Gateway code. 
Even when PABOOST is active, the values of the RSSI received vary from -110s to -50s or -60s when the SNR changes from positive to negatives values.
According to the SX1272 datasheet the formulas used in the calculations are right, so I have no idea what is wrong. 
Any help will be appreciated, and hopefully someone can point me in the right direction to solve the problem.
Thanks in advance for your assitance!

CongducPham, 2017-10-04:
The RSSI variation does not mean that something's wrong. If at small 
ranges the SNR is high (i.e. 8 or 9) then the PABOOST settings is 
correct. If you see SNR=-6 for instance at 2m then it most likely means 
that the PABOOST setting is not correct.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

richoco, 2017-10-13:
Hello again, thank you for the reply.

The values of SNR at small ranges are correct, as you mentioned, all close to 6 or 7, proving that the PABOOST is active. The problem I have is when the SNR fluctuates between positives (or zero) and negatives values. 

Here is an example: this is data taken from 300m, and you can see how, when SNR goes from negative to zero, the RSSI  drastically changes, making it impossible to compare with similar scenarios in which this problem does not occur.

RSSI | SNR
--     |   --
-50   | -1
-114 | 0
-50   | -1
-52   | -2
-52   | -2
-53   | -3
-53   | -3
-113 | 0

Regards,

Ricardo

CongducPham, 2017-11-10:
Yes, this is an issue that we have to look at when SNR is negative. Formula is Semtech's doc is not very clear. Will check that.

richoco, 2017-11-12:
Thank you for the follow up. I will keep checking the topic in case of any update.

Regards, 

Ricardo

CongducPham, 2017-11-23:
We changed the way the RSSI is computed. Should be more now. You can get the new SX1272.cpp file.

richoco, 2017-11-27:
Nice, thank you for the quick fix! As soon as I can, I will try the new file and report back.

Regards, 

Ricardo