CongducPham/LowCostLoRaGw #146: MongoDB cloud is not working.

ethicstechOSS, 2017-10-04:
Hi. I am using latest sdcard image uploaded by you on last Monday. I am using two node devices and RPI3 as gateway. I enabled local MongoDB writing in clouds.json and disabled all other clouds. I can see the log that I am receiving the datas from nodes on the gateway . However, I am not able to see any entry in MongoDB web page. The Display sources row also empty. 

I get into mongo and checked the db.ReceivedData.find().pretty() as per this guide, https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-NewCloud.md#support-of-mongodb-as-a-cloud-declaration. But it return nothing. 
I was able to see the datas on MongoDB in my older image. But not getting in this latest sdcard image. 

TIA,
Prakash.

noesberger, 2017-10-04:
Hi, I manually installed the software and I'm also not able to write data to mongodb.
```
pi@loragw01:~/lora_gateway $ sudo ./lora_gateway | python ./post_processing_gw.py

post_processing_gw.py found an alert_conf section
Parsing cloud declarations
[u'python CloudMongoDB.py']
Parsed all cloud declarations
post_processing_gw.py got cloud list:
[u'python CloudMongoDB.py']
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got encrypted cloud list:
[]
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got LoRaWAN encrypted cloud list:
[]
Starting thread to report gw status
2017-10-04 23:16:16.045235
post status: gw ON, lat 46.887891 long 7.77651

Current working directory: /home/pi/lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
```

to send data I'm using for test a feather 32u4 and the example:
Arduino_LoRa_Ping_Pong.ino
and I change the code to use no-ack version:
```
      // e = sx1272.sendPacketTimeoutACK(DEFAULT_DEST_ADDR, message, r_size);
      // this is the no-ack version
      e = sx1272.sendPacketTimeout(DEFAULT_DEST_ADDR, message, r_size);
```

and change the String to send:
`  r_size=sprintf((char*)message, "TC/22.5");`

It's receiving data but not writing anything to mongodb:
```
-----------------------------------------------------
Check for downlink requests 2017-10-04T23:20:56
NO NEW DOWNLINK ENTRY
-----------------------------------------------------
--- rxlora. dst=1 type=0x10 src=8 seq=92 len=7 SNR=-17 RSSIpkt=-58 BW=125 CR=4/5 SF=12
2017-10-04T23:20:57.941702
rcv ctrl pkt info (^p): 1,16,8,92,7,-17,-58
splitted in:  [1, 16, 8, 92, 7, -17, -58]
(dst=1 type=0x10(DATA) src=8 seq=92 len=7 SNR=-17 RSSI=-58)
rcv ctrl radio info (^r): 125,5,12
splitted in:  [125, 5, 12]
(BW=125 CR=5 SF=12)
rcv timestamp (^t): 2017-10-04T23:20:57.940

got first framing byte
--> got data prefix
TC/22.5
```
It created the DB messages, but no collection.

Thanks for your help

CongducPham, 2017-10-05:
To enable cloud upload (MongoDB is considered as a cloud) you have to 
use a prefix:

\!TC/22.5

Please, look at the Arduino_LoRa_Simpl_temp instead of ping-pong.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

noesberger, 2017-10-05:
Thanks for your help. It's sometimes so simple.
With the prefix \! it's working.

closed 2017-10-14 by ethicstechOSS