CongducPham/LowCostLoRaGw #147: rn2903

sebaz143, 2017-10-06:
Hi,

I am using SX1272 transceiver and a raspberry pi3 with your os, i have succesfully transmited data from a node with a sx1272 transceiver in 900 Mhz band. i have another lora node with a rn2903, but i can't receive data in the gateway from this node. 

I have ran the lora_gateway with the --freq 908.3 argument.

I have the rn2903 conneced to a usb host in my computer and i have the run this commands:

sys get ver
sys get hweui

mac reset
mac set devaddr 00000006
mac set nwkskey 2B7E151628AED2A6ABF7158809CF4F3C
mac set appskey 2B7E151628AED2A6ABF7158809CF4F3C
mac set appkey 00112233445566778899AABBCCDDEEFF
mac set pwridx 5
mac set dr 0
mac set adr off
mac set ar off
mac set sync 12

mac set ch status 0 off ..... mac set ch status 71 off (all but the 30 channel, this one is for 908.3 Mhz)

mac save
mac join abp
mac tx uncnf 1 01

CongducPham, 2017-10-08:
Please, read the info on LoRaWAN compatibility here:

https://github.com/CongducPham/LowCostLoRaGw/blob/master/gw_full_latest/README-aes_lorawan.md

regards,

sebaz143, 2017-10-10:
I'm still not able to send packets from my rn2903 to the low cost lora gateway with the sx1272 lora radio in the 900 Mhz band. I read above document and did everything it says. Th only thing that i get is a Receive error 2 and then the radio module restart. But after the second or third restarting, the gateway get freeze and i have to run the lora_gateway manually.

I hope you can help me, I'm out of ideas. I have changed all the parameters with no success.

This is my configuration:

# **GATEWAY_CONF.JSON**

{
        "radio_conf" : {
                "mode" : 3,
                "bw" : 125,
                "cr" : 5,
                "sf" : 10,
                "ch" : 2,
                "freq" : 907.5
        },
        "gateway_conf" : {
                "gateway_ID" : "00000027EB2DB98E",
                "ref_latitude" : "my_lat",
                "ref_longitude" : "my_long",
                "wappkey" : false,
                "raw" : true,
                "aes" : true,
                "log_post_processing" : true,
                "log_weekly" : false,
                "dht22" : 0,
                "dht22_mongo": false,
                "downlink" : 0,
                "status" : 600,
                "aux_radio" : 0
        },
        "alert_conf" : {
                "use_mail" : false,
                "contact_mail" : "the_contact_mail_address,another_contact_mail_a$
                "mail_from" : "a_gmail_address",
                "mail_server" : "smtp.gmail.com",
                "mail_passwd" : "your_passwd",
                "use_sms" : false,
                "pin": "0000",
                "contact_sms":["+33XXXXXXXXX","+33XXXXXXXXX"],
                "gammurc_file":"/home/pi/.gammurc"
        }
}

# **RUN THE GATEWAY WITH THIS COMMAND**

sudo ./lora_gateway --mode 3 --cr 5 --sf 10 --bw 125 --raw --sw 34 --freq 907.5 | python post_processing_gw.py

**RN2903 CONFIGURATION**

mac reset
mac set devaddr 00000006
mac set nwkskey 2B7E151628AED2A6ABF7158809CF4F3C
mac set appskey 2B7E151628AED2A6ABF7158809CF4F3C
mac set appkey 00112233445566778899AABBCCDDEEFF
mac set pwridx 8
mac set dr 0
mac set adr off
mac set ar off
mac set sync 34
mac set ch status 0 off ..... mac set ch status 71 off (all but the 10 channel, this one is for 907.5 Mhz)
mac set ch drrange 10 0 0 
mac save
mac join abp
mac tx uncnf 1 01

# **GATEWAY OUTPUT**

pi@raspberrypi:~/lora_gateway $ sudo ./lora_gateway --mode 3 --cr 5 --sf 10 --bw 125 --raw --sw 34 --freq 907.5 | python post_processing_gw.py
raw output from low-level gateway. post_processing_gw will handle packet format
enable local AES decryption
post_processing_gw.py found an alert_conf section
Parsing cloud declarations
[u'python CloudMongoDB.py']
[u'python CloudMongoDB.py', u'python CloudMySQL.py']
Parsed all cloud declarations
post_processing_gw.py got cloud list:
[u'python CloudMongoDB.py', u'python CloudMySQL.py']
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got encrypted cloud list:
[]
Parsing cloud declarations
Parsed all cloud declarations
post_processing_gw.py got LoRaWAN encrypted cloud list:
[]
Starting thread to report gw status
2017-10-11 00:04:48.437357
post status: gw ON, lat my_lat long my_long

Current working directory: /home/pi/lora_gateway
SX1272 detected, starting.
...
**********Power ON: state 0
Default sync word: 0x12
Set sync word to 0x34
LoRa sync word: state 0
LoRa CR 5: state 0
LoRa SF 10: state 0
LoRa BW 125: state 0
Frequency 907.500000: state 0
Set LoRa power dBm to 14
Power: state 0
Get Preamble Length: state 0
Preamble Length: 8
LoRa addr 1: state 0
Raw format, not assuming any header in reception
SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Low-level gw status ON
## Receive error 2
Resetting radio module
SX1272 detected, starting.
...
Setting power ON: state 0
LoRa CR 5: state 0
LoRa SF 10: state 0
LoRa BW 125: state 0
Frequency 907.500000: state 0
Set LoRa power dBm to 14
Power: state 0
Get Preamble Length: state 0
Preamble Length: 8
LoRa addr 1: state 0
Raw format, not assuming any header in reception
SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
Low-level gw status ON

MZourob, 2018-08-15:
Hi @sebaz143 ,

I am trying to exactly do what you have described in this thread. I have a RN2903 module and I am not able to receive anything on the gateway that is a RPi3 with a Dragino LoRa/GPS hat. Have you figured out a solution for this?

Thank you.

MZourob, 2018-08-16:
Hi @CongducPham  ,

I am trying to exactly do what @sebaz143 was trying to do. I have a RN2903 module and I am not able to receive anything on the gateway that is a RPi3 with a Dragino LoRa/GPS hat. Have you figured out a solution for this?

Thank you very much for the help.

CongducPham, 2018-08-16:
We have updated how CRC are handled, can you check again with latest version from github?

regards,

MZourob, 2018-08-16:
Thank you @CongducPham for your reply. I have just installed the Jessie image like 2 days ago into my Pi, so I guess I have the latest version, is not that correct?

CongducPham, 2018-08-17:
OK, yes, you should have the latest version, please check the git version and the installed version by looking at the 2 files in /home/pi: git-VERSION.txt and VERSION.txt

MZourob, 2018-08-20:
I have confirmed the version but still no use. I still cannot receive from the LoRa Mote. I am not sure but the LoRa Mote operates in the region from 902 to 923 Mhz and the Hat I have it seems is operating at 868 Mhz. So I would assume that there is a frequency mismatch, correct?