CongducPham/LowCostLoRaGw #150: Compile Error Arduino_LoRa_Generic_Sensor with DHT22.cpp module

kkalbaugh, 2017-10-31:
When I try to compile the  Arduino Generic Sensor I get the following compile errors:

```
DHT22.cpp: In constructor 'DHT22::DHT22(uint8_t)':

DHT22.cpp:69: error: cannot convert 'volatile uint32_t* {aka volatile long unsigned int*}' to 'volatile uint8_t* {aka volatile unsigned char*}' in assignment

     _baseReg = portInputRegister(digitalPinToPort(pin));

              ^

DHT22.cpp: In member function 'DHT22_ERROR_t DHT22::readData()':

DHT22.cpp:108: error: 'cli' was not declared in this scope

   cli();

       ^

DHT22.cpp:110: error: 'sei' was not declared in this scope

   sei();

       ^
```
Tried searching for answers but my C++ skills are too weak at this point to wrap my head around converting integers.  

Thanks, Kirby

kkalbaugh, 2017-10-31:
I probably should add that I'm using a Adafruit M0 LoRa Feather.

CongducPham, 2017-11-23:
We haven't fully tested the sketch with the FeatherM0 and it looks like the DHT lib has some issue on this board. You can avoid the compilation error by commenting:

//#include "DHT22_Humidity.h"
//#include "DHT22_Temperature.h"

and also:

  //sensor_ptrs[1] = new DHT22_Temperature("TC", IS_NOT_ANALOG, IS_CONNECTED, low_power_status, (uint8_t) 3, (uint8_t) 9);
  //sensor_ptrs[2] = new DHT22_Humidity("HU", IS_NOT_ANALOG, IS_CONNECTED, low_power_status, (uint8_t) 3, (uint8_t) 9);

change the number of sensors and their index, otherwise just copy the first sensor as follows, just for test:

  sensor_ptrs[1] = new LM35("tc", IS_ANALOG, IS_CONNECTED, low_power_status, (uint8_t) A0, (uint8_t) 8);
  sensor_ptrs[2] = new LM35("tc", IS_ANALOG, IS_CONNECTED, low_power_status, (uint8_t) A0, (uint8_t) 8);

Then move all DHT22* file from the Arduino_LoRa_Generic_Sensor, close the sketch, open it again and compile.

We will try to debug the DHT22 sensor for the FeatherM0 later.

regards,