CongducPham/LowCostLoRaGw #151: Problems with communication from Arduino device to Pi gateway

emilvi, 2017-10-31:
Thank you for this really impressive software and documentation

I have set up a Pi3 with your image, and configured the setup. I also have a Ardiuno Uno. Both are set up with new inAir9B.
I have have set up PABOOST on and off, and different LORAMODE (1,2,and 3)

Here is the output from both Pi and Ardiuno which seems OK for me. I had problems with compiling Ardiuno_lora_temp, but after delete sx1272.h and let it use the files in the library it worked OK (I have also testet with copying the library files to the same folder, and It compiled OK.) but no connections between devices. I have also changed the library Arduino AVR boards to 1.6.11

I have followed the cabling from both video on Youtube and documents, and it seems OK

Any suggestions?

_Following last lines of ../Dropbox/LoRa-test/post-processing.log. CTRL-C to return
Current UTC date is
Tue 31 Oct 19:42:22 UTC 2017
2017-10-31T20:41:49.959170> Parsing cloud declarations
2017-10-31T20:41:49.959470> Parsed all cloud declarations
2017-10-31T20:41:49.959774> post_processing_gw.py got LoRaWAN encrypted cloud list:
2017-10-31T20:41:49.960089> []
2017-10-31T20:41:49.960391> Starting thread to report gw status
2017-10-31T20:41:49.960727> 2017-10-31 20:41:49.938777
2017-10-31T20:41:49.961044> post status: gw ON, lat my_lat long my_long
2017-10-31T20:41:50.942552>
2017-10-31T20:41:50.943008> Current working directory: /home/pi/lora_gateway
2017-10-31T20:41:50.943349> SX1276 detected, starting.
2017-10-31T20:41:50.943669> SX1276 LF/HF calibration
2017-10-31T20:41:50.943987> ...
2017-10-31T20:41:50.944293> **********Power ON: state 0
2017-10-31T20:41:50.944675> Default sync word: 0x12
2017-10-31T20:41:50.945029> LoRa mode 3
2017-10-31T20:41:50.945352> Setting mode: state 0
2017-10-31T20:41:50.945665> Channel CH_10_868: state 0
2017-10-31T20:41:50.945981> Use PA_BOOST amplifier line
2017-10-31T20:41:50.946300> Set LoRa power dBm to 14
2017-10-31T20:41:50.946615> Power: state 0
2017-10-31T20:41:50.946928> Get Preamble Length: state 0
2017-10-31T20:41:50.947247> Preamble Length: 8
2017-10-31T20:41:50.947563> LoRa addr 1: state 0
2017-10-31T20:41:50.947876> SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
2017-10-31T20:41:51.339497> Low-level gw status ON_

_LoRa temperature sensor, extended version
ATmega328P detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Get back previous sx1272 config
Using packet sequence number of 11
Forced to use default parameters
Using node addr of 6
Using idle period of 1
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
Reading 0
Temp is 0.00
Sending \!#1#TC/0.00
Real payload size is 12
--> CAD duration 131
OK1
--> waiting for 1 CAD = 16
--> CAD duration 131
OK2
LoRa pkt size 16
LoRa pkt seq 11
LoRa Sent in 581
LoRa Sent w/CAD in 712
Packet sent, state 0
Switch to power saving mode
Successfully switch LoRa module in sleep mode_

emilvi, 2017-11-05:
Sorry I have mistaken the inAir9 module, what I have is inAir9 (not 9B which I tought), but still I have problems. I have commented out the PABOOST in arduino, and changed the radio.makefile file.
Is there any other settings that have to be set?

_pi@raspberrypi:~/lora_gateway $ cat radio.makefile
#options are: -DPABOOST -DMAX_DBM=v
CFLAGS=-DMAX_DBM=14_

I still have problems with sending I wonder what the RSSI=125 means

_Simple LoRa ping-pong with the gateway
ATmega328P detected
SX1276 detected, starting
SX1276 LF/HF calibration
...
Setting Mode: state 0
Setting Channel: state 0
Setting Power: state 0
Setting node addr: state 0
SX1272 successfully configured
--> CAD duration 549
OK1
--> waiting for 1 CAD = 62
--> CAD duration 549
OK2
## RSSI value is -125 ##

--> RSSI -125
Sending Ping
## Packet set and written in FIFO ##
## Packet to send: 
Destination: 1
Packet type: 24
Source: 8
Packet number: 0
Packet length: 8
Data: Ping
##
wait for ACK_

and output from raspberry

_pi@raspberrypi:~/lora_gateway $ sudo ./lora_gateway
SX1276 detected, starting.
SX1276 LF/HF calibration
...
^$**********Power ON: state 0
^$Default sync word: 0x12
^$LoRa mode 1
^$Setting mode: state 0
^$Channel CH_10_868: state 0
^$Set LoRa power dBm to 14
^$Power: state 0
^$Get Preamble Length: state 0
^$Preamble Length: 8
^$LoRa addr 1: state 0
^$SX1272/76 configured as LR-BS. Waiting RF input for transparent RF-serial bridge
^$Low-level gw status ON_

CongducPham, 2017-11-10:
Hi, sorry to ask, but did you recompile after changing the radio.mkefile, then reboot?

emilvi, 2017-11-11:
Yes I have recompiled, with same result
I have ordered new inAir9 to check if there is any HW problems. And also new boards

emilvi, 2017-11-13:
It seems that when connecting inAir9 VSWITCH (VS) to 3.3 V on the Arduino it worked right away. Should the VS be connected to any other pin instead of 3.3V?

CongducPham, 2017-11-13:
Strange, I never use this pin.

regards,