CongducPham/LowCostLoRaGw #152: Весь код в багах, утечки памяти

juvf, 2017-11-07:
Автор!!! Что вы делаете? Прежде чем писать на с++, нужно выучить с++. 

Решил сделать свой код на основе этого. Выкинуть всё лишнее, дополнить своим. Что я вижу

/* Prints data to the serial port as human-readable ASCII text followed
 * by a carriage retrun character '\r' and a newline character '\n' */
void SerialPi::println(float f, int precission){
    const char *str1="%.";
    char * str2;
    char * str3;
    char * message;
    sprintf(str2,"%df",precission); //печатаем в str2, но под str2 не выделена память, BAG!!!
    asprintf(&str3,"%s%s",str1,str2); //выделяем память str3 чтобы влезло str1 и str2 и печатаем в str3 
    sprintf(message,str3,f); //печатаем в message, но память под message не выделана. BAG!!!

    const char *newline="\r\n";
    char * msg = NULL;
    asprintf(&msg,"%s%s",message,newline);//выделяем память msg чтобы влезло messageи и newline и печатаем в msg
    unistd::write(sd,msg,strlen(msg)); //отправляем в порт
}//Выход!!! функцией asprintf() было дважды выделение памяти в куче. в описании функции написанно

avlad, 2017-11-10:
Sure, this is not a single problem with this repository ... but, what the reason write to foreign guy in Russian? :)

juvf, 2017-11-10:
Привет. Во первых, возможно ты русскоговорящий. Во вторых мой английский слабый. Боюсь, что если я напишу на английском, ты вообще ни чего не поймёшь. Google/yandex переводчики удаляют языковые барьеры. Думаю для тебя не проблема перевести моё сообщение.