CongducPham/LowCostLoRaGw #153: Image support for RPI Zero W or OrangePi 2G IOT Board [Features]

ethicstechOSS, 2017-11-08:
Any plan to support for RPI Zero/W or OrangePi 2G IOT boards?. I already tried for OPI 2G IOT board since it is low cost with 2G onboard module. However, the board is not yet stable, then I am moving to RPI Zero/W board. Share any info if any one already tried for RPI Zero/W.

TIA,
Prakash.

ethicstechOSS, 2017-11-08:
I tried the same img file on Zero W. The Raspbian booted without any issues. However, the log shows that the Radio module is not up. Looks like some SPI connection issue, but the "ls -al /dev/spi*" shows two SPI devices. Should I recompile the WiringPi for ZERO W board to enable the SPI?.

TIA,
Prakash

CongducPham, 2017-11-08:
What log are you referring to? How did you connect the radio module? 
Maybe you are running the lora_gateway program twice as on boot an 
instance is running. See here:

https://github.com/CongducPham/LowCostLoRaGw/tree/master/gw_full_latest#use-cmdsh-to-interact-with-the-gateway

*IMPORTANT NOTICE*: Do not launch a new gateway instance with an 
existing one as there will be conflict on the SPI bus.

regards,

C. Pham.

-- 
|------------ Congduc PHAM - Professor --------------------------|
|  LIUPPA - Equipe T2I              |         U.P.P.A. Pau       |
|  http://liuppa.univ-pau.fr/       |   http://www.univ-pau.fr   |
|----------------------------------------------------------------|
|UPPA, LIUPPA laboratory, UFR Sciences et Techniques             |
|Avenue de l'Université - BP 1155                                |
|64013 PAU CEDEX, FRANCE                                         |
|                                                                |
|phone:  [33] (0) 5 59 40 75 94                                  |
|fax:    [33] (0) 5 59 40 76 54                                  |
|Congduc.Pham@univ-pau.fr       http://www.univ-pau.fr/~cpham    |
|----------------------------------------------------------------|

ethicstechOSS, 2017-11-09:
Hi. I tried the same SPI communication with RPI3 & Zero W. I confirmed that the LoRa module is working fine on RPI3; But the getting this log from Zero W and I confirmed the lora_gateway instance not executed twice,

*******Log starts here*********************
Current UTC date is
Thu  9 Nov 04:59:24 UTC 2017
2017-11-09T05:58:58.097028> post_processing_gw.py found an alert_conf section
2017-11-09T05:58:58.103018> Parsing cloud declarations
2017-11-09T05:58:58.110163> [u'python CloudThingSpeak.py']
2017-11-09T05:58:58.111328> Parsed all cloud declarations
2017-11-09T05:58:58.119096> post_processing_gw.py got cloud list: 
2017-11-09T05:58:58.124882> [u'python CloudThingSpeak.py']
2017-11-09T05:58:58.126065> Parsing cloud declarations
2017-11-09T05:58:58.127156> Parsed all cloud declarations
2017-11-09T05:58:58.128251> post_processing_gw.py got encrypted cloud list: 
2017-11-09T05:58:58.129364> []
2017-11-09T05:58:58.130454> Parsing cloud declarations
2017-11-09T05:58:58.131551> Parsed all cloud declarations
2017-11-09T05:58:58.158838> post_processing_gw.py got LoRaWAN encrypted cloud list: 
2017-11-09T05:58:58.159958> []
2017-11-09T05:58:58.161048> Starting thread to report gw status
2017-11-09T05:58:58.192411> 2017-11-09 05:58:58.097866
2017-11-09T05:58:58.193585> post status: gw ON, lat my_lat long my_long
2017-11-09T05:58:59.112515> 
2017-11-09T05:58:59.113713> Current working directory: /home/pi/lora_gateway
*******Log ends here*********************
The above log never shows the LORA module details.

Any kind of assistance appreciated. 

TIA,
Prakash.

ethicstechOSS, 2017-11-10:
Tried the image on 3 Zero W boards. But facing the same problem. I am also seeing that few others also facing SPI issues on the Zero W board. Just curious to know if you have been tested the OS image file on Zero W board?.

TIA,
Prakash

CongducPham, 2017-11-10:
Seems that you didn't receive my last mail:

---

OK, I think I see what happen. I guess you took the SD card image from the web. This image has the gateway program compiled for RPI2&RPI3. For the RPI0 it is similar to RPI1 so just type:

make lora_gateway

and you will have the correct version of the program.

---
I will make it clearer in the README.
regards,

ethicstechOSS, 2017-11-10:
Thanks for your input. I will follow your instruction and report you. But it looks like the same. Because, the SPI enabled, SPI devices are available and I already tried the sample SPI loop program given by the RPI. It was working, but the gateway program dont deliver the output as expected.

CongducPham, 2017-11-10:
It should work when recompiling, because you SPI loop program was compiled on the RPIzero.

CongducPham, 2017-11-10:
Done. See https://github.com/CongducPham/LowCostLoRaGw#features. Thanks for reporting this issue.

closed 2017-11-11 by ethicstechOSS