CongducPham/LowCostLoRaGw #155: issues on distance coverage with PABOOST.

ethicstechOSS, 2017-11-23:
Hi,
I am using SX1276 based radio from ACSIP. I am only getting upto 500m in open space even when I enabled PABOOST and increased the dB upto 20. Do you have any suggestions or specific parameter to increase the coverage distance b/w node & gateway?. BTW, I am using 2.5dB antenna on both side. 

TIA.

CongducPham, 2017-11-23:
Do you have a link to the radio module?

ethicstechOSS, 2017-11-23:
After 500m, I lost the connection b/w the node & gateway.

CongducPham, 2018-04-04:
Well, I've never the radio module you mentioned, can you try using an RFM95?

ethicstechOSS, 2018-04-04:
Yes, I will try with RFM95 and let you know.

Just curious to know if my above steps enough to get good distance (I am not in the situation to get one like you. Anything less than 5km is fine for me) or If I missed anything here to configure at firmware level?.

TIA.

ethicstechOSS, 2018-04-06:
Hi. Tested with RFM95W. I can able to receive datas for upto 2.5kms clearly with above same setup. After 100m-200m, got some junk values and then disconnected. I was sending sample temperature data for every 5 sec. 

Like to know if I have any other configuration to modify and achieve more than 2.5kms. 

Appreciate your patient and continuously supporting me.

ethicstechOSS, 2018-04-06:
"clear LOS condition," - You mean  plain open space b/w the node & gateway, no building/tree in between?. In my case, I have only couple of 2 floor buildings & trees b/w the node & gateway. How about increasing the gateway antenna in more height?.

CongducPham, 2018-04-10:
Sure, this is always better. There is a tradeoff between a long cable (higher) and the attenuation that you will get. A clear LOS is nothing in the Fresnel zone. If you have obstacle, even few, then it is not clear LOS, but probably tractable.